libimobiledevice/libimobiledevice #921: Any GUI for libimobiledevice?

keg3w, 2020-03-08:
Is there any open-source GUI for libimobiledevice on any platform?

mexmer, 2020-07-29:
libimobile is library not program, therefore it can't have GUI.

you write programs on top of it.

michi7801, 2020-08-13:
OK, but is there any known GUI that uses this library? 
Would be a charm for noobs who are using Linux for office work. 
I'm sure there would be contributors for such a project!

demhademha, 2020-08-13:
> OK, but is there any known GUI that uses this library? 
> 
> Would be a charm for noobs who are using Linux for office work. 
> 
> I'm sure there would be contributors for such a project!
If you  explain what type of work, that'd be excellent

michi7801, 2020-08-29:
I am not sure what you mean. I just wanted to know, if any GUI exists, or if there are any efforts to create one?

michi7801, 2020-09-03:
I know that. I was never said that this library should have or has a GUI. Again I just want to know if there are any projects developing a GUI with this library as a base?

P3nguin-M, 2021-10-11:
> I am not sure what you mean. I just wanted to know, if any GUI exists, or if there are any efforts to create one?

3uTools most likely uses it and it's a GUI.

JoshAtticus, 2022-09-22:
> > I am not sure what you mean. I just wanted to know, if any GUI exists, or if there are any efforts to create one?
> 
> 3uTools most likely uses it and it's a GUI.

Windows only