libimobiledevice/libimobiledevice #922: ERROR: Could not connect to lockdownd: Invalid HostID (-21)

jhihn, 2020-03-10:
This error appears when a host which has previously been trusted by the device has had its trust reset (Settings: General: Reset: Reset Location & Privacy) and is plugged in again. 

The work-around is to delete the existing trust file in the lockdown directory and re-pair.

It would be nice if this could happen automatically.

mexmer, 2020-03-12:
you can manually call unpair

lmamane, 2020-09-22:
I had this issue; I don't remember ever resetting the trust. I didn't even have to unpair, calling "idevicepair pair" worked.

rootkittt, 2023-01-16:
where can i find the trust file?

ryanjkemper, 2024-04-18:
Ran into the issue today and can confirm running idevicepair pair does resolve this issue when it occurred for me.

I see the security advantage of keeping the current functionality and forcing a re-pair.  I am building re-try logic into my script that utilizes this library for backups.  I do like @RJVB's [idea to at least improve the error message](https://github.com/libimobiledevice/libimobiledevice/issues/1394#issuecomment-1898532474) by suggesting to re-pair the device.

Veshurik, 2024-06-15:
Had same problem, went to Settings - General - Reset - Reset Location & Privacy, then reconnect to USB, pop-up appeared with trusting on device, press "Trust", and finally it is working again.

Unforunately, button "Re-pair" in Sideloadly doesn't working, because it still writes same error Invalid HostID (-21), so you need to manually delete trust settings on device. 
It's still wasn't fixed (normal description of error), as I can see, because now it v0.50.3.

mexmer, 2024-06-17:
@Veshurik on machine do 
`idevicepair unpair `
this command will remove pairing information from usbmuxd (even if it's not valid)
then run
`idevicepair pair`
this should show trust dialog

Veshurik, 2024-06-17:
> on machine do

How I can do such commands if I don't know where to enter them?

mexmer, 2024-06-18:
linux terminal, where else would you enter commands?

Veshurik, 2024-06-18:
Linux? I don't have Linux at all. Only Windows

mexmer, 2024-06-18:
then in commandline of windows, if you have compiled libimobile, you have these compiled too.