libimobiledevice/libimobiledevice #930: The Python iconModDate value lists 1989 instead of 2020

dnicolson, 2020-04-05:
The following code outputs dates from 1989 instead of 2020. The C function `sbservices_get_icon_state` returns correct values.

macOS 10.15.4 (19E266)
iOS 13.4 (17E255)
Python 2.7.16 (system Python)
Cython 0.29.16

```python
import imobiledevice

idevice = imobiledevice.iDevice()
ld = imobiledevice.LockdownClient(idevice, 'sb')
sbport = ld.start_service(imobiledevice.SpringboardServicesClient)
sbc = imobiledevice.SpringboardServicesClient(idevice, sbport)
icon_state = sbc.icon_state

for s in icon_state:
  for i in s:
      print i['iconModDate']
```

```
>>> for s in icon_state:
...   for i in s:
...       print i['iconModDate']
...
1989-03-26 09:16:24.886524
```

nikias, 2020-04-13:
This is an issue in the cython bindings for libplist. It needs a offset "MACH_EPOCH" which has to be added to the actual raw value that's stored in the plist data. I will push an update for libplist.

nikias, 2020-04-13:
I fixed this with libplist commit https://github.com/libimobiledevice/libplist/commit/e7355a408a305d0a9aba7c30bbd5a33d978a03c7

closed 2020-04-14 by dnicolson