libimobiledevice/libimobiledevice #931: has no member named 'file_handle' 91

Disx, 2020-04-13:
Help me please!

makepkg error:

afc.c: In function 'afc_client_new_with_service_client': afc.c:91:12: error: 'afc_client_private' {aka 'struct afc_client_private'} has no member named 'file_handle' 91 | client_loc->file_handle = 0; | ^~ afc.c:92:12: error: 'afc_client_private' {aka 'struct afc_client_private'} has no member named 'lock' 92 | client_loc->lock = 0; | ^~ make[2]: [Makefile:565: afc.lo] Error 1 make[2]: Leaving directory '/opt/libimobiledevice-git/src/libimobiledevice/src' make[1]: [Makefile:450: all-recursive] Error 1 make[1]: Leaving directory '/opt/libimobiledevice-git/src/libimobiledevice' make: *** [Makefile:380: all] Error 2 ==> ERROR: A failure occurred in build(). Aborting...

huyuwei1996, 2020-04-13:
我今天弄了半天，一直安装也是这个报错。
可以尝试切换上个commit，我编译安装正常。
xxx为你的用户名
cd /Users/xxx/Library/Caches/Homebrew/libimobiledevice--git
git checkout c0576790378daa45d43335d44c5caff04c270c94
brew install gnutls
brew install libgcrypt
./autogen.sh --disable-openssl
make
make install
我这里是直接安装了一些别的包，没有使用自动安装的openssl@1.1编译

nikias, 2020-04-13:
@Disx please pull latest. I forgot to update afc.c aswell in my previous commit.

Disx, 2020-04-13:
> @Disx please pull latest. I forgot to update afc.c aswell in my previous commit.

Thanks, Installed...

closed 2020-04-13 by Disx