libimobiledevice/libimobiledevice #934: Skip Setup assistant using libi

tausift0, 2020-04-24:
There are programs like 3uTools and iMazing that are able to skip the setup wizard after restoring the device and go straight to the Home screen. I have been trying to do that in libi but there seems to be no way to do so. Can anyone guide me how we can achieve that in libi? I have to restore a lot of devices and I don't want to manually do it all. Please help!!

Dantee296, 2020-04-24:
@tausift0  this is possible

tausift0, 2020-04-24:
I have tried a lot. I am not able to do it. How is it possible?

tausift0, 2020-04-25:
@nikias 
Please can you confirm if this is possible or not.

Kanong017, 2020-04-29:
I have a way to do it. Contact me: 185252060@qq.com

dmartin-bit, 2020-04-30:
I am trying to do the same thing at the moment. So if you find a solution it would be great if you could post it here.

tausift0, 2020-04-30:
Yes please post it here the solution. I tried contacting on the mail too but got no reply.

truonggiang0710, 2020-05-27:
Hi @tausift0 ,
Do you have any solutions for this ?

Dantee296, 2020-05-27:
@tausift0 i thought u ll figure it our your self 
email me , and lmme know why you need it

jduranc, 2020-05-27:
help_request ++;

Hi guys, 

I have some time researching this feature also, so, if you get some information I really appreciate it if you share it.

truonggiang0710, 2020-05-28:
Hi @Dantee296 ,
I send you the email but you still do not reply !
If you can public this code, please reply my email.

andreipetrashkevich, 2020-05-28:
as I understand apple has broken something in it, but when I looked at the method in setup app that uses ForceNoBuddy. the method is the same as previous implementation for iOS <13.3.

andreipetrashkevich, 2020-05-28:
3uTools and iMazing use the iTunes framework to reach this aim.

tausift0, 2020-05-28:
> Hi @tausift0 ,
> Do you have any solutions for this ?
> 

I still have not found any solution for this and those who claim to know are not willing to share it here. They want you to purchase the 'trick' from them.
I will surely share it here if I find something.

a-gunter, 2021-01-13:
I've also been trying to look into this recently...

Here's what I've gathered so far:

- You can write values to the com.apple.purplebuddy domain, and querying info from com.apple.PurpleBuddy will reflect the values that you've written after a restart on the device (it's strange how you need to write to purplebuddy, but then the actual important values seem to come from PurpleBuddy. Writing directly to PurpleBuddy domain doesn't seem to accomplish much).
- You can write language / locale via com.apple.international
- You can skip all setup steps using something like Apple Configurator to supervise the device
- Even after all of these steps, there's one last obstacle which is the iOS Terms and Conditions. Configurator does not seem to give the option to skip this screen, and I'm not sure if PurpleBuddy holds any information about this one. I don't see a difference in PurpleBuddy before / after this step is completed. I think Configurator may be able to skip this screen if the device is enrolled in DEP? Not sure though. Although this isn't really related to libimobiledevice.

Has anybody else been able to discover information regarding this? It seems like the iOS Terms and Conditions is the real blocker.

clemAwrld, 2021-01-17:
pls am still hoping for a solution on this.......for ios 9.3.5

Nutr1t07, 2021-03-12:
@a-gunter May I ask that how to write to the com.apple.purplebuddy domain? Is there a way to do it with iPhone not activated yet?

a-gunter, 2021-03-13:
Not sure if activation is a requirement or not, but that should be easy to test. I'm on mobile currently so I can't say for sure. 

As for writing to the domain, start a session with lockdown and then specify the domain, key, and value when writing to it. You can look at a simple tool like idevicename to get an idea.

```
// setting device name
lerr = lockdownd_set_value(lockdown, NULL, "DeviceName", plist_new_string(argv[0]));
```

Instead of NULL, you would specify com.apple.purplebuddy, and instead of "DeviceName", you would specify the key that you want to set. You can get an idea of the possible keys by running `ideviceinfo -q com.apple.PurpleBuddy` after the device has completed setup. Values can vary between bools, strings, arrays, and dicts so you'll have to pay attention to what the key types are when you're trying to set them

a-gunter, 2021-03-13:
And for what it's worth, there's a new iOS 14+ method of skipping the setup screens that involves installing a mobileconfig profile, but the device must be supervised beforehand or else the profile cannot be installed (I think because the payload type includes com.apple.managed?). So regardless, you will need some method to supervise the device like through Apple Configurator. Not sure if 3utools or something similar has supervision capabilities, or if libimobiledevice can achieve the same thing.

Nutr1t07, 2021-03-14:
@a-gunter Thanks for your patient explanations! I forgot that this is an C library... Now I know how to write values to nodes in `com.apple.PurpleBuddy`. But I come across a new problem: I can't remove nodes from `com.apple.PurpleBuddy`(#1131). It is maybe an irreversible operation to skip Setup assistant.

truonggiang0710, 2022-08-25:
@Nutr1t07 @a-gunter Can we set ForceNoBuddy = true to skip setup assistant now?

Nutr1t07, 2022-08-27:
> @Nutr1t07 have you achieved it or not. As 3utools and imagzing not longer supported to skip the setupwizard

no. i didn't achieve it. and i haven't bothered with this stuff for quite a long time

Rodionp13, 2023-05-18:
> And for what it's worth, there's a new iOS 14+ method of skipping the setup screens that involves installing a mobileconfig profile, but the device must be supervised beforehand or else the profile [...]

Hi @a-gunter , could you please provide some more details about mobileconfig profile which is used for  skipping the setup screens ? For today, I've faced a problem with restore-backup on iOS16.4+ devices. After restore device appear on 'HelloScreen'.

windowalter, 2023-08-10:
I have already skip langauge and country selection step but i cannot skip wifi next button part.
I have tried many things but not getting results.

joseluiskuuc, 2023-10-10:
Has anyone managed to skip the setup steps?

windowalter, 2023-10-20:
> 3uTools and iMazing use the iTunes framework to reach this aim.

how it uses them