libimobiledevice/libimobiledevice #938: new -fsanitize=address breaks build under gcc 8.1.0/mingw

elekven-m, 2020-04-30:
Commit 8bd8250336694b0a68546fa01417ef40748c84a4 breaks my build under gcc 8.1.0/mingw because configure.ac adds "-fsanitize=address" to AS_COMPILER_FLAGS but does not define the proper symbols for the linker to link to the asan library, leading to many undefined symbols such as 
__asan_report_store4 and __asan_report_load4. See https://stackoverflow.com/questions/50610323/undefined-symbol-asan-option-detect-stack-use-after-return and https://stackoverflow.com/questions/50621054/getting-undefined-symbol-asan-memset-when-trying-to-use-clang-address-sanitiz

nikias, 2020-04-30:
Yes that was a mistake. I will remove it.

nikias, 2020-04-30:
Fixed with e13e4863edb38b5e67b40b9404738fa40920b6bf.

closed 2020-04-30 by nikias