libimobiledevice/libimobiledevice #941: Different problems with openssl vs gnutls

philipl, 2020-05-01:
Hi.

I've been trying to find a combination of library versions that allows Ubuntu 20.04 to connect to my Ipad running iOS 13 and transfer files. What I've found is that things are broken one way with openssl and another way with gnutls.

The minimum case that I'm testing is rebuilding libimobiledevice from the latest git.

If I build with gnutls, I can plug in, establish trust and browse around just fine, but any attempt to copy a file to the device makes no progress. Sits there at 0% forever.

If I build with openssl, usbmuxd reports that the stored pairing info is invalid and forces repairing every time I plug in. After re-pairing, the connection works and I can transfer files.

But right now I don't know how to get everything to work at the same time.

The ubuntu package is built against gnutls and exhibits the hanging file transfer problem (so, Ubuntu is broken out of the box).

I could easily imagine this has to do with how gnutls and/or openssl have been built by ubuntu. I did not try rebuilding those libraries.

nikias, 2020-05-01:
I am not sure what you are doing but at least for me with OpenSSL on ubuntu it will keep the trust in place. Have you made sure that usbmuxd is handling this correctly? Mind that usbmuxd also links against libimobiledevice so if you didn't care about usbmuxd at all it will probably be the one that comes with ubuntu which links against an older libimobiledevice which in turn might cause the issues with your newer library version causingit to re-pair every time you plug in. Does it ask you to pair as soon as you plug it in, or only when you try to run a command (like ideviceinfo) ?

philipl, 2020-05-01:
I have tried with both a rebuilt usbmuxd and the stock one with no difference. I get `lockdown error -5` and then it re-pairs, every time I plug it in.

nikias, 2020-05-01:
Can you make sure the pairing record gets stored correctly? Check in /var/lib/lockdown

philipl, 2020-05-01:
It's creating the pairing record and the contents look reasonable - it then deletes it after it says the record is bad, then writes out a new one with various different values, which I think is unsurprising.

qmfrederik, 2020-06-15:
#974 / f56a65f7bc08de0cd9bc5c1e4683b1d0c671b767 should fix this. Could you test this on your end?

philipl, 2020-06-15:
That fixes the openssl problem. Thanks! As long as I'm building my own library, the gnutls problem now doesn't block anything.

seb128, 2020-06-23:
Ubuntu is using the gnutls backend for the same reason than Debian, licensing issue between openssl and the GPL, to fix the default packages in Ubuntu we will need to fix the gnutls backend

qmfrederik, 2020-06-23:
Well, do you have some exact steps to reproduce the issue you're seeing with libimobiledevice 1.3 on Ubuntu 20.04 when using GnuTLS?

GnuTLS seems to be correctly configured (there's a call to `gnutls_priority_set_direct` in `idevice.c` which allows most ciphers, see 72643b2b83990b9cf97cc84b285b30763d44a72d), so it must be something else.

philipl, 2020-06-23:
The exact steps for me were:
- Have an IOS 13 device
- Plug it in
- Establish trust if necessary (no difference if trust is new or pre-existing)
- Try and copy a file to the device using Nautilus (using gvfsd-afc which uses libimobiledevice)
- File copy copies zero bytes, with no log entries noted for usbmuxd or gvfsd-afc
- Basic navigation (browsing app directories, etc) in Nautilus works.

Repeating these steps with an openssl build of the same libimobiledevice and no other changes works correctly,

qmfrederik, 2020-06-23:
Hmm, I don't know much about Nautilus or how it interacts with libimobiledevice. Can you reproduce the hang with any of the libimobiledevice utils?

mexmer, 2020-06-24:
problem is not in gnutls, ios doesn't encrypt AFC connection (except on jailbroken phones, where most current implementation sets afc2 to encrypted)
this is different problem, if files have 0 size and give no error, more likely it's bad handling of socket connection with afc.

nikias, 2020-06-24:
@mexmer that's not entirely true, since iOS 13 AFC will enforce SSL even via USB.

philipl, 2020-06-24:
The problem doesn't occur with backups - files transfer fine there. Is there a libimobiledevice family tool that exercises afc? Or some other afc tool you'd like me to test with?

philipl, 2020-06-24:
I tried using ifuse and that can't even list directory contents with gnutls
`ls: cannot open directory '.': Input/output error`

The equivalent openssl library build works fine (although file transfers are much much slower than in nautilus)

mexmer, 2020-06-25:
@nikias i just checked AFC with service start, and no ... it's not ssl enabled.
i have ios 13.5 on my iphone SE 2020 and 13.5.1 on ipad air 3.

some services are ssl enabled, but AFC over USB is not

request
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
<key>Label</key>
<string>MOBILedit Apple Client version 1.0</string>
<key>Request</key>
<string>StartService</string>
<key>Service</key>
<string>com.apple.afc</string>
</dict>
</plist>
```
response
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
<key>Port</key>
<integer>51158</integer>
<key>Request</key>
<string>StartService</string>
<key>Service</key>
<string>com.apple.afc</string>
</dict>
</plist>
```

mexmer, 2020-06-25:
btw. jailbroken iphone 8 with ios 13.3 (using checkm8te)

start afc
request
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
<key>Label</key>
<string>Apple Client version 1.0</string>
<key>Request</key>
<string>StartService</string>
<key>Service</key>
<string>com.apple.afc</string>
</dict>
</plist>
```
response
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
<key>Port</key>
<integer>51126</integer>
<key>Request</key>
<string>StartService</string>
<key>Service</key>
<string>com.apple.afc</string>
</dict>
</plist>
```
start afc2
request
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
<key>Label</key>
<string>Apple Client version 1.0</string>
<key>Request</key>
<string>StartService</string>
<key>Service</key>
<string>com.apple.afc2</string>
</dict>
</plist>
```
response
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
<key>EnableServiceSSL</key>
<true/>
<key>Port</key>
<integer>51129</integer>
<key>Request</key>
<string>StartService</string>
<key>Service</key>
<string>com.apple.afc2</string>
</dict>
</plist>
```

mexmer, 2020-06-25:
tbh. i'm not aware about apple fixing SSL for AFC ... it's like common knowledge, that apple didn't do that
but for example house_arrest service (which also utilise AFC) is using SSL

qmfrederik, 2020-06-25:
> Is there a libimobiledevice family tool that exercises afc? Or some other afc tool you'd like me to test with?

@philipl I think you can give idevicecrashreport a try. That should excercise afc. If you specify the `-k` flag it should keep the crash reports on the device (so you can run it multiple times & have afc copy over those files).

philipl, 2020-06-25:
@qmfrederik Yep. I was able to reproduce with `idevicecrashreport`. With the openssl build, the files copy with no problem. With the gnutls build, there is a long delay, and then it says:

`Unable to open device file './<something>.ips' (32). Skipping...`

followed by multiple lines of:

`Failed to read information for './<something>.ips'. Skipping...`

qmfrederik, 2020-12-04:
@philipl Quick question - do you have `gvfs-backends` installed? Does uninstalling `gvfs-backends` fix this for you? May be related to #1037

philipl, 2020-12-04:
@qmfrederik I'm using nautilus and gvfs to access my device - so it doesn't matter if it works without `gvfs-backends`.

nikias, 2020-12-04:
@philipl I think there is some logical issue in the implementation that handles gnutls. I need to revise the code that does the ssl reads.

qmfrederik, 2020-12-07:
@philipl I get that, though it may be easier to pinpoint the problem we know it consistently reproduces when gvfs-afc is running.

Regarding gnutls: I've seen this on desktop Ubuntu 20.10, and as far as I know, [libimobiledevice6](https://packages.ubuntu.com/groovy/libimobiledevice6) in Ubuntu [uses OpenSSL](http://changelogs.ubuntu.com/changelogs/pool/main/libi/libimobiledevice/libimobiledevice_1.3.0-5/changelog):

>  [ Yves-Alexis Perez ]
>  * build against OpenSSL
>
> -- Yves-Alexis Perez [email]  Wed, 19 Aug 2020 17:32:59 +0200

philipl, 2020-12-07:
Yes @qmfrederik , Ubuntu updated their packages to use OpenSSL and that's unblocked me on a day to day basis. I can certainly find some time to do my own gnutls build and test again if that will help. But is there anyone that gnutls is working for for file transfers?