libimobiledevice/libimobiledevice #943: Unexpected error while compiling in c++ with g++ and makefile

AirKetchPLAYZ, 2020-05-03:
I get this error on compile:
```
In file included from main.cpp:3:
/usr/local/include/libimobiledevice/diagnostics_relay.h:179:133: error: expected ‘,’ or ‘...’ before ‘class’
  179 | diagnostics_relay_error_t diagnostics_relay_query_ioregistry_entry(diagnostics_relay_client_t client, const char* name, const char* class, plist_t* result);
      |                                                                                                                                     ^~~~~
make: *** [<builtin>: main.o] Error 1
"make all" terminated with exit code 2. Build might be incomplete.
```
with this makefile
```
OBJS	= main.o
SOURCE	= main.cpp
OUT	= output
CC	 = g++
FLAGS	 = -g -c -Wall
LFLAGS	 = 

all: $(OBJS)
	$(CC) -g $(OBJS) -o $(OUT) $(LFLAGS) 

clean:
	rm -f $(OUT)

run: $(OUT)
	./$(OUT)
```
Whats wrong?

nikias, 2020-05-03:
It's a c++ name collision (using class as a c function parameter name). I will push a fix.

nikias, 2020-05-03:
Pull latest, I fixed this with commit 21f1da655c019d4f6b2eab06c6cd4a42a724d51d.

closed 2020-05-03 by nikias