libimobiledevice/libimobiledevice #944: undefined reference to all the functions in this lib I am calling

AirKetchPLAYZ, 2020-05-04:
I have this script 
```c++
#include "libimobiledevice/libimobiledevice.h"
#include "libimobiledevice/lockdown.h"
#include "libimobiledevice/diagnostics_relay.h"

#include <iostream>

using namespace std;

int main() {
	printf("start");
	idevice_t device = NULL;
	diagnostics_relay_client_t diagnostics_client = NULL;
	lockdownd_service_descriptor_t service = NULL;
	lockdownd_error_t ret = LOCKDOWN_E_UNKNOWN_ERROR;
	lockdownd_client_t lockdown_client = NULL;

	const char *udid = NULL;

	ret = lockdownd_client_new_with_handshake(device, &lockdown_client, "idevicediagnostics");
	idevice_new(&device, udid);
	ret = lockdownd_start_service(lockdown_client, "com.apple.mobile.diagnostics_relay", &service);
	diagnostics_relay_client_new(device, service, &diagnostics_client);
	diagnostics_relay_restart(diagnostics_client, DIAGNOSTICS_RELAY_ACTION_FLAG_WAIT_FOR_DISCONNECT);
	return 0;
}
```
But I get these errors:
```
main.cpp:(.text+0x71): undefined reference to `lockdownd_client_new_with_handshake'
/usr/bin/ld: main.cpp:(.text+0x87): undefined reference to `idevice_new'
/usr/bin/ld: main.cpp:(.text+0x9e): undefined reference to `lockdownd_start_service'
/usr/bin/ld: main.cpp:(.text+0xb8): undefined reference to `diagnostics_relay_client_new'
/usr/bin/ld: main.cpp:(.text+0xc9): undefined reference to `diagnostics_relay_restart'
```
I have /usr/local/include on my include path and I'm using c++
```

AirKetchPLAYZ, 2020-05-04:
I realised that in my /usr/local/include/libimobiledevice/ there are only .h files and no .c files

nikias, 2020-05-04:
Can you stop opening issues for things that are not actual bugs IN the library? You have general issues that you can just search for on your favorite search engine.
Yes, include path does not include .c files, you installed the library and are supposed to link against it, so you need to adapt your linker flags adding `-limobiledevice` to it, possibly also `-L/usr/local/lib`.
Besides you are supposed to use < > instead of " " for your includes in your sources.

Again: This is not a support forum for your compiler/linker/dev env problems so do your research before opening issues and make sure they are actual issues within the library.

closed 2020-05-04 by nikias