libimobiledevice/libimobiledevice #947: ideviceinfo com.apple.mobile.battery - over WiFi?

JCsplash, 2020-05-08:
Is it possible to use ideviceinfo to get battery info **without usb connection** for ios devices with WiFi sync turned on?

JCsplash, 2020-05-08:
ideviceinfo -u DID -q com.apple.mobile.battery seems to work only when iPhone / iPad is unlocked. Wondering if there's a way to get battery info even if the screens locked?

closed 2020-05-08 by JCsplash

nikias, 2020-05-08:
The wifi connection is not permanent. It only works when the device wakes up every now and then.

JCsplash, 2020-05-09:
You're right Nikias but the wait is at most a couple minutes in my case so it isn't bad. Thanks for all you do!

Do you think it'll ever be possible to read Apple Watch's battery level through ideviceinfo? Maybe battery level can go from Watch → iPhone → Custom Bonjour (?) Sync Service → ideviceinfo / mac. Would this be possible?

nikias, 2020-05-09:
@JCsplash for Apple watch we need to go through companion proxy. I think this is possible.

JCsplash, 2020-05-09:
@nikias Would you mind elaborating a bit? Would this approach require an apple watch app and/or iPhone app?

Right now I plan to do it the brute force way :(
1. Get battery level on custom Apple Watch App (via background tasks)
2. Send charge to iPhone companion app (via watch connectivity)
3. Send charge to Firebase from iPhone 
4. Mac reads from firebase.

If somehow we could achieve Apple Watch → Mac directly that’d be huge. But not sure what exactly you’re thinking of.

nikias, 2020-05-10:
I will post an addition to the library in the upcoming days and then you implement it for your needs. No need for an app. Just stay put.

JCsplash, 2020-05-11:
Thank you @nikias! I just finished the app above which works but is overkill and slow (watch connectivity only gives me at most 2 updates an hr).

I can’t wait to try your solution. I'm already using ideviceinfo to get ipad / iphone charge so this would be perfect. If you’d be down to share your thought process along the way, I’d really appreciate that as well!

JCsplash, 2020-05-12:
Also @nikias I found the code below that tries this com.apple.companion_proxy approach. 

https://gist.github.com/matteyeux/04df80e5368d99a001b7

Maybe this could help?

nikias, 2020-05-18:
@JCsplash check my latest commit 0268777212847afd7734cb280115aea70dc81c92 and this:
https://gist.github.com/nikias/ebc6e975dc908f3741af0f789c5b1088

JCsplash, 2020-05-19:
@nikias thank you for making this, it looks great! I’m not exactly sure how i can use it though. Would you mind sharing some more instructions? Is there a command i can call like ideviceinfo?

nikias, 2020-05-19:
@JCsplash compile and run my test program `comptest`  that I posed in my previous comment.

JCsplash, 2020-05-19:
[image]

@nikias I'm getting " error: no matching function for call to 'idevice_new_with_options'". I pretty sure i have the latest code. Any thoughts on what to do here?

nikias, 2020-05-19:
You must have an older version installed somewhere.

JCsplash, 2020-05-19:
I get a different error when using gcc to compile. Are you able to successfully compile/link by just doing "gcc comptest.c"?  Or are there extra steps i need to take to link comptest to libimobiledevice files?

[image]

JCsplash, 2020-05-19:
This works amazing. Thank you so much @nikias !  Is there any way to make it run a bit faster like idevice_id or ideviceinfo? Right now it takes about 3-4s.

Here's what I did to get it working (if anyone else needs help)

**Compiling Code**

```
gcc <INSERT_PATH_TO_COMPTEST> -L<INSERT_PATH_TO DIRECTORY_WITH_LIBIMOBILEDEVICE.A> -limobiledevice  -L<INSERT_PATH_TO_DIRECTORY_WITH_LIBPLIST.A> -lplist -o <INSERT_YOUR_FILE_NAME>
```

Move executable to /usr/local/bin so it can be used globally (Optional) 

**Running Code**

```
<INSERT_YOUR_FILE_NAME> <INSERT_IPHONE_UDID>
```

nikias, 2020-05-19:
you need latest libimobiledevice and libplist too and then you can just compile it with
```
gcc -o comptest comptest.c -limobiledevice -lplist
```

nikias, 2020-05-19:
It doesn't run faster, this is how Apple implemented it unfortunately. Like every time you query a key it closes the connection from the device side...

JCsplash, 2020-05-19:
No worries, this is great. Do you know if apple pencil also uses companion_proxy to connect to iPad or if that's just for watches?

nikias, 2020-05-19:
Just watches. Appel Pencil uses Bluetooth LE.

JCsplash, 2020-05-19:
@nikias  Sounds good. What do you think about adding a timeout to comptest? Do you thinnk it's necessary?

nikias, 2020-05-20:
Do whatever you want with it, it's just a proof of concept.

JCsplash, 2020-05-20:
Sounds good. Thank you so much @nikias !

JCsplash, 2020-05-21:
Hey @nikias is there a "BatteryIsCharging" or "ExternalChargeCapable" value for Apple Watch? I tried using get_value_from_watch but just got back "Unexpected error occurred". 

I'm wondering if those fields even exist?

nikias, 2020-05-21:
`BatteryIsCharging` definitely exists. See my updated example in the gist.

nikias, 2020-05-21:
Also please update libplist again, I had a name mismatch between header file and public interface function.

JCsplash, 2020-05-21:
@nikias Did you push your libplist changes? I still see "bbde6a4" as the last libplist commit made 6 days ago?

nikias, 2020-05-21:
my bad, now I did

JCsplash, 2020-05-21:
Just updated libplist, compiled, and ```BatteryIsCharging``` works great! Thanks @nikias !

JCsplash, 2020-05-23:
Hey @nikias , quick question. Do you know if WiFi networks can have WiFi sync disabled by their administrators? I realized WiFi sync works on my WPA2 Personal Network but not on the WPA2 Enterprise network at my school. Is this typical?

nikias, 2020-05-23:
Maybe the admins are dropping certain packet types. WiFi sync uses mDNS for device discovery.

JCsplash, 2020-10-05:
Hey @nikias ! By any chance, do you know how [ScreenShieldKit](https://screenshieldkit.com) is able to prevent screenshots on iOS? I’ve read it has something to do with converting UILabels to DRM Video. You’re the only person (besides the creators of ScreenShieldKit) who I think would know the answer to this. I couldn’t find your email so decided just to reach out here! (I can delete this comment later).