libimobiledevice/libimobiledevice #953: make on macOS 10.15 failed

dreamerblue, 2020-05-18:
Found this dependency from https://github.com/RemoteDebug/remotedebug-ios-webkit-adapter.

Then I installed `libimobiledevice` by command `brew install --HEAD libimobiledevice` and failed with error `Undefined symbols for architecture`.

OS: macOS 10.15.4
Clang Version: Apple clang version 11.0.3 (clang-1103.0.32.59)

```
Updating Homebrew...
==> Cloning https://git.libimobiledevice.org/libimobiledevice.git
Cloning into '/Users/User/Library/Caches/Homebrew/libimobiledevice--git'...
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
==> ./autogen.sh
==> ./configure --prefix=/usr/local/Cellar/libimobiledevice/HEAD-0268777_6 --without-cython --enable-debug-code
==> make install
Last 15 lines from /Users/User/Library/Logs/Homebrew/libimobiledevice/03.make:
libtool: compile:  clang -DHAVE_CONFIG_H -I. -I.. -I../include -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/libusbmuxd/HEAD-5e3c636/include -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/openssl@1.1/1.1.1g/include -g -O2 -c companion_proxy.c  -fno-common -DPIC -o .libs/companion_proxy.o
libtool: compile:  clang -DHAVE_CONFIG_H -I. -I.. -I../include -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/libusbmuxd/HEAD-5e3c636/include -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/openssl@1.1/1.1.1g/include -g -O2 -c syslog_relay.c  -fno-common -DPIC -o .libs/syslog_relay.o
libtool: compile:  clang -DHAVE_CONFIG_H -I. -I.. -I../include -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/libusbmuxd/HEAD-5e3c636/include -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/openssl@1.1/1.1.1g/include -g -O2 -c preboard.c -o preboard.o >/dev/null 2>&1
libtool: compile:  clang -DHAVE_CONFIG_H -I. -I.. -I../include -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/libusbmuxd/HEAD-5e3c636/include -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/openssl@1.1/1.1.1g/include -g -O2 -c companion_proxy.c -o companion_proxy.o >/dev/null 2>&1
libtool: compile:  clang -DHAVE_CONFIG_H -I. -I.. -I../include -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/libusbmuxd/HEAD-5e3c636/include -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/openssl@1.1/1.1.1g/include -g -O2 -c syslog_relay.c -o syslog_relay.o >/dev/null 2>&1
/bin/sh ../libtool  --tag=CC   --mode=link clang -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/libusbmuxd/HEAD-5e3c636/include   -I/usr/local/Cellar/libplist/2.1.0/include  -I/usr/local/Cellar/openssl@1.1/1.1.1g/include  -g -O2   -L/usr/local/Cellar/libplist/2.1.0/lib -lplist -L/usr/local/Cellar/libusbmuxd/HEAD-5e3c636/lib -lusbmuxd  -L/usr/local/Cellar/openssl@1.1/1.1.1g/lib -lssl -lcrypto  -version-info 6:0:0 -no-undefined   -o libimobiledevice.la -rpath /usr/local/Cellar/libimobiledevice/HEAD-0268777_6/lib idevice.lo service.lo property_list_service.lo device_link_service.lo lockdown.lo afc.lo file_relay.lo notification_proxy.lo installation_proxy.lo sbservices.lo mobile_image_mounter.lo screenshotr.lo mobilesync.lo mobilebackup.lo house_arrest.lo mobilebackup2.lo misagent.lo restore.lo diagnostics_relay.lo heartbeat.lo debugserver.lo webinspector.lo mobileactivation.lo preboard.lo companion_proxy.lo syslog_relay.lo ../common/libinternalcommon.la
libtool: link: clang -dynamiclib  -o .libs/libimobiledevice.6.dylib  .libs/idevice.o .libs/service.o .libs/property_list_service.o .libs/device_link_service.o .libs/lockdown.o .libs/afc.o .libs/file_relay.o .libs/notification_proxy.o .libs/installation_proxy.o .libs/sbservices.o .libs/mobile_image_mounter.o .libs/screenshotr.o .libs/mobilesync.o .libs/mobilebackup.o .libs/house_arrest.o .libs/mobilebackup2.o .libs/misagent.o .libs/restore.o .libs/diagnostics_relay.o .libs/heartbeat.o .libs/debugserver.o .libs/webinspector.o .libs/mobileactivation.o .libs/preboard.o .libs/companion_proxy.o .libs/syslog_relay.o   -Wl,-force_load,../common/.libs/libinternalcommon.a  -L/usr/local/Cellar/libplist/2.1.0/lib -L/usr/local/Cellar/libusbmuxd/HEAD-5e3c636/lib -L/usr/local/Cellar/openssl@1.1/1.1.1g/lib -lssl -lcrypto -lusbmuxd -lplist  -g -O2   -install_name  /usr/local/Cellar/libimobiledevice/HEAD-0268777_6/lib/libimobiledevice.6.dylib -compatibility_version 7 -current_version 7.0 -Wl,-single_module
Undefined symbols for architecture x86_64:
  "_plist_string_val_compare", referenced from:
      _compproxy_get_device_registry in companion_proxy.o
      _compproxy_get_value_from_registry in companion_proxy.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [libimobiledevice.la] Error 1
make: *** [install-recursive] Error 1

READ THIS: https://docs.brew.sh/Troubleshooting

Please create pull requests instead of asking for help on Homebrew's GitHub,
Discourse, Twitter or IRC.
```

bigzhu, 2020-05-18:
Same issue

nikias, 2020-05-18:
You need to update libplist.

bigzhu, 2020-05-19:
> You need to update libplist.

works for me :) Thanks @nikias

kuluo, 2020-05-19:
> You need to update libplist.

This works for me. run 
`
brew unlink libplist
brew install --HEAD libplist
`

dreamerblue, 2020-05-19:
@nikias @kuluo Thanks. It works for me!

closed 2020-05-19 by dreamerblue

yangyang72, 2020-05-20:
> > You need to update libplist.
> 
> This works for me. run
> `
> brew unlink libplist
> 
> brew install --HEAD libplist
> `

thank you very much~