libimobiledevice/libimobiledevice #954: How Can I enable Wifi Sync on iOS device via libimobiledevice?

sprhawk, 2020-05-18:
Is there any example for it? or any document about it?

Or how can I investigate it?

JCsplash, 2020-05-19:
@sprhawk For MacOS Catalina, you need to check "Show this iPhone when on WiFi". 

For OS Mojave, go to itunes and check "Sync iPhone over WiFi"

If you've checked the boxes but your mac still can't detect your device wirelessly, you might need to reset network settings on your device. This thread might help.

https://discussions.apple.com/thread/250876485

sprhawk, 2020-05-19:
> @sprhawk For MacOS Catalina, you need to check "Show this iPhone when on WiFi".
>
> For OS Mojave, go to itunes and check "Sync iPhone over WiFi"
>
> If you've checked the boxes but your mac still can't [...]

Hi, thanks for reply. But what I asked is via libimobiledeivce, not by iTunes.

MarSoft, 2021-11-12:
Sometines it will help if you `idevicepair unpair` and then `idevicepair pair` the device.