libimobiledevice/libimobiledevice #955: unable to install libimobiledevice --HEAD

EdulaSuryaKiranReddy, 2020-05-18:
Error on running a command 'brew install --HEAD libimobiledevice'.
Env: 
Mac OS Mojave (10.14.6)
Xcode (11.3.1)

Please find the below logs.

Xxxx:~ user$ brew install --HEAD libimobiledevice
Updating Homebrew...
==> Cloning https://git.libimobiledevice.org/libimobiledevice.git
Updating /Users/user/Library/Caches/Homebrew/libimobiledevice--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at 0268777 Add support for com.apple.companion_proxy
==> ./autogen.sh
==> ./configure --prefix=/usr/local/Cellar/libimobiledevice/HEAD-0268777_6 --without-cython --enable-debug-code
==> make install
Last 15 lines from /Users/user/Library/Logs/Homebrew/libimobiledevice/03.make:
libtool: compile:  clang -DHAVE_CONFIG_H -I. -I.. -I../include -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/libusbmuxd/HEAD-5e3c636/include -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/openssl@1.1/1.1.1g/include -g -O2 -c mobileactivation.c -o mobileactivation.o >/dev/null 2>&1
libtool: compile:  clang -DHAVE_CONFIG_H -I. -I.. -I../include -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/libusbmuxd/HEAD-5e3c636/include -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/openssl@1.1/1.1.1g/include -g -O2 -c debugserver.c -o debugserver.o >/dev/null 2>&1
libtool: compile:  clang -DHAVE_CONFIG_H -I. -I.. -I../include -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/libusbmuxd/HEAD-5e3c636/include -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/openssl@1.1/1.1.1g/include -g -O2 -c preboard.c -o preboard.o >/dev/null 2>&1
libtool: compile:  clang -DHAVE_CONFIG_H -I. -I.. -I../include -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/libusbmuxd/HEAD-5e3c636/include -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/openssl@1.1/1.1.1g/include -g -O2 -c syslog_relay.c -o syslog_relay.o >/dev/null 2>&1
libtool: compile:  clang -DHAVE_CONFIG_H -I. -I.. -I../include -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/libusbmuxd/HEAD-5e3c636/include -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/openssl@1.1/1.1.1g/include -g -O2 -c companion_proxy.c -o companion_proxy.o >/dev/null 2>&1
/bin/sh ../libtool  --tag=CC   --mode=link clang -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/libusbmuxd/HEAD-5e3c636/include   -I/usr/local/Cellar/libplist/2.1.0/include  -I/usr/local/Cellar/openssl@1.1/1.1.1g/include  -g -O2   -L/usr/local/Cellar/libplist/2.1.0/lib -lplist -L/usr/local/Cellar/libusbmuxd/HEAD-5e3c636/lib -lusbmuxd  -L/usr/local/Cellar/openssl@1.1/1.1.1g/lib -lssl -lcrypto  -version-info 6:0:0 -no-undefined   -o libimobiledevice.la -rpath /usr/local/Cellar/libimobiledevice/HEAD-0268777_6/lib idevice.lo service.lo property_list_service.lo device_link_service.lo lockdown.lo afc.lo file_relay.lo notification_proxy.lo installation_proxy.lo sbservices.lo mobile_image_mounter.lo screenshotr.lo mobilesync.lo mobilebackup.lo house_arrest.lo mobilebackup2.lo misagent.lo restore.lo diagnostics_relay.lo heartbeat.lo debugserver.lo webinspector.lo mobileactivation.lo preboard.lo companion_proxy.lo syslog_relay.lo ../common/libinternalcommon.la  
libtool: link: clang -dynamiclib  -o .libs/libimobiledevice.6.dylib  .libs/idevice.o .libs/service.o .libs/property_list_service.o .libs/device_link_service.o .libs/lockdown.o .libs/afc.o .libs/file_relay.o .libs/notification_proxy.o .libs/installation_proxy.o .libs/sbservices.o .libs/mobile_image_mounter.o .libs/screenshotr.o .libs/mobilesync.o .libs/mobilebackup.o .libs/house_arrest.o .libs/mobilebackup2.o .libs/misagent.o .libs/restore.o .libs/diagnostics_relay.o .libs/heartbeat.o .libs/debugserver.o .libs/webinspector.o .libs/mobileactivation.o .libs/preboard.o .libs/companion_proxy.o .libs/syslog_relay.o   -Wl,-force_load,../common/.libs/libinternalcommon.a  -L/usr/local/Cellar/libplist/2.1.0/lib ### -L/usr/local/Cellar/libusbmuxd/HEAD-5e3c636/lib -L/usr/local/Cellar/openssl@1.1/1.1.1g/lib -lssl -lcrypto -lusbmuxd -lplist  -g -O2   -install_name  /usr/local/Cellar/libimobiledevice/HEAD-0268777_6/lib/libimobiledevice.6.dylib -compatibility_version 7 -current_version 7.0 -Wl,-single_module
**Undefined symbols for architecture x86_64:
  "_plist_string_val_compare", referenced from:
      _compproxy_get_device_registry in companion_proxy.o
      _compproxy_get_value_from_registry in companion_proxy.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [libimobiledevice.la] Error 1
make: *** [install-recursive] Error 1**

nikias, 2020-05-18:
You need to update libplist.

EdulaSuryaKiranReddy, 2020-05-19:
Thanks @nikias 
updating libplist to HEAD worked for me.

closed 2020-05-19 by EdulaSuryaKiranReddy