libimobiledevice/libimobiledevice #956: Problems when install libimobiledevice from brew

alxjandroszlv, 2020-05-20:
I have a problem installing lib with brew using HEAD. Any advice?

```
brew install --HEAD libimobiledevice
```

Undefined symbols for architecture x86_64:
  "_plist_string_val_compare", referenced from:
      _compproxy_get_device_registry in companion_proxy.o
      _compproxy_get_value_from_registry in companion_proxy.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

make[1]: *** [libimobiledevice.la] Error 1
make: *** [install-recursive] Error 1

nikias, 2020-05-20:
You have to update libplist.

VladislavBukariev, 2020-05-20:
> You have to update libplist.
Worked for me
    brew uninstall libplist
    brew install libplist --HEAD
    brew install libimobiledevice --HEAD

alxjandroszlv, 2020-05-20:
Thank you so much!
The problem has been solve

closed 2020-05-20 by alxjandroszlv