libimobiledevice/libimobiledevice #957: Can't make libimobiledevice on Manjaro Linux 20.0.1

andrsn0w, 2020-05-20:
In file included from /usr/include/string.h:495,
                 from idevicecrashreport.c:29:
In function ‘strncpy’,
    inlined from ‘afc_client_copy_and_remove_crash_reports’ at idevicecrashreport.c:153:4:
/usr/include/bits/string_fortified.h:106:10: warning: ‘__builtin_strncpy’ specified bound depends on the length of the source argument [-Wstringop-overflow=]
  106 |   return __builtin___strncpy_chk (__dest, __src, __len, __bos (__dest));
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
idevicecrashreport.c: In function ‘afc_client_copy_and_remove_crash_reports’:
idevicecrashreport.c:152:17: note: length computed here
  152 |    int newlen = strlen(list[k]) - 7;
      |                 ^~~~~~~~~~~~~~~
  CCLD     idevicepair
  CCLD     idevicename
/usr/bin/ld: ../src/.libs/libimobiledevice.so: undefined reference to `plist_string_val_compare'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:682: idevice_id] Error 1
make[2]: *** Waiting for unfinished jobs....
/usr/bin/ld: ../src/.libs/libimobiledevice.so: undefined reference to `plist_string_val_compare'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:722: ideviceinfo] Error 1
/usr/bin/ld: ../src/.libs/libimobiledevice.so: undefined reference to `plist_string_val_compare'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:726: idevicename] Error 1
/usr/bin/ld: ../src/.libs/libimobiledevice.so: undefined reference to `plist_string_val_compare'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:734: idevicepair] Error 1
make[2]: Leaving directory '/var/tmp/pamac-build-andrias/libimobiledevice-git/src/libimobiledevice/tools'
make[1]: *** [Makefile:450: all-recursive] Error 1
make[1]: Leaving directory '/var/tmp/pamac-build-andrias/libimobiledevice-git/src/libimobiledevice'
make: *** [Makefile:380: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...

I tried to build https://aur.archlinux.org/packages/libimobiledevice-git but this error Ioccurs when i download libimobiledevice directly from git and try to make it too.
I could use libimobiledevice that comes with Manjaro itself, but i want to use idevicerestore, so i need libimobiledevice-git

mexmer, 2020-05-20:
you need also updated libplist

andrsn0w, 2020-05-20:
> you need also updated libplist

Thanks! It worked. Now i can use idevicerestore again with my Manjaro Linux operating system!

closed 2020-05-20 by andrsn0w