libimobiledevice/libimobiledevice #958: idevicebackup2 doesn't restore much of user data

lmamane, 2020-05-20:
I'm restoring a backup made on:
iPad Pro 9.7" Model A1674 with iOS 13.4.1
to
iPad Pro 11" Model A2230 with iOS 13.4.1

The source iPad does _not_ have an iCloud account configured, so my understanding is that "cloud backup" must be off. Running "idevicebackup2 -u OLD_IPAD_ID cloud off"
says:

Started "com.apple.mobilebackup2" service on port 50066.
Negotiated Protocol Version 2.1
ErrorCode 1: No account exists (MBErrorDomain/1)
Could not disable cloud backup.

But much of the user data is _not_ restored:
 * No photo/video is restored. But I know the photos are in the backup, because I can find them by looking for large JPEG files in the backup. For example, "54/544c8e844fff39f1e2c670175218a8a96eda10d7" is one, a1/a1d1edf094c4c6956761b1bfc4b2c69cb165edc8 is another, etc.
 * The data of the "Windy Maps" (id cz.seznam.windymaps), the downloaded maps, is not restored.
 * The audio and video files that are in VLC for iOS (id org.videolan.vlc-ios) on the old iPad are not restored.
 * All the downloaded PDFs in "Books" (the Apple app, id com.apple.iBooks) are not restored. The actual libraries (categories) are recreated in books, but they are all empty.

lmamane, 2020-05-20:
Also not restored:
* Safari bookmarks (favourites)
* Mail accounts configuration

Have been restored:
* Notes "on my iPad"
* EasyZIP (app id net.webdia.eazyzip) files

lmamane, 2020-05-20:
Also restored:
* Numbers (the Apple app, id com.apple.Numbers) files
* Files in "Downloads"

lmamane, 2020-05-20:
Also not restored:
* Website & App Passwords
* Setting of what Siri can and cannot access (went back to Siri can access everything)
* what app / system service is allowed access to location
* what app is allowed background refresh and not
* what app is allowed notifications and not
* Enabled dictionaries
* Sounds setting "keyboard clicks"
* "do not disturb" setting "repeated calls"

Were restored:
* Safari default search engine
* Safari "page zoom on" website liste
* Safarie "website data" in "Advanced" of Safari Settings
* Timezone settings
* Enabled keyboards (languages)

nikias, 2020-05-20:
You just ran `cloud off`? Or did you actually run restore? For the restore operation you need to attach the new iPad and run
```
idevicebackup2 -s OLD_ID -u NEW_ID --system --settings restore path/to/backupdir
```

nikias, 2020-05-20:
Also I really recommend that you enable backup encryption and make a new backup from the old device before you try to restore. Otherwise you lose WiFi passwords, credentials, app tokens, etc.

lmamane, 2020-05-20:
I ran the restore operation, with the same command line you gave except without "--system", and I added "--copy". Since I was restoring to a different model of iPad, "--system" sounded like a bad idea.

It restored a bunch of stuff (listed above), but didn't restore other things (listed above).

For the VLC videos, they were not in the backup, that is because of misconfiguration of VLC, it was configured to _not_ include media library. So that is not a libimobiledevice problem, reason understood.

I found the Safari bookmarks in a sqlite3 database in the old backups, so they are definitely in the backup. Just not being restored. Same with the photos.

nikias, 2020-05-20:
Just try what I said. Make an encrypted backup, and then restore with `--system --settings`. I have used this before and got all my data back.

lmamane, 2020-05-20:
OK, thanks for the advice. Started doing that now, full encrypted backup in a new backup directory. Will try to restore that after.

lmamane, 2020-05-20:
Yes, that worked much better. Mail accounts, Safari bookmarks (but not opened tabs), etc are now restored. Much thanks.

However, it didn't restore PDFs that are in collections in the "Book" app. Do you have an idea of the reason for that, and how to solve it?

nikias, 2020-05-20:
Well usually if something doesn't show up after restore it is not included in the actual backup. It might be that something has to be added specifically for this specific "domain". I just checked on my old iPhone X which had a few PDFs, and they are indeed not on my new iPhone 11 Pro. So something must be missing here.

lmamane, 2020-05-21:
For completeness, I also had a DRM-free epub file (not downloaded from Apple Books service, but imported from a file). Also not restored, probably for the same reasons.

Small remark, I used --skip-apps on the last restore. The apps were redownloaded automatically anyway. <shrug>

nikias, 2020-05-22:
Ok I think the problem with the iBooks is that it doesn't restore it properly. The backup should contain it. Can you check in your backup directory/UDID/Info.plist if it has a `iBooks Data 2` entry? No need to post it here, but just wondering if you see it. I assume this has to be manually restored after the process is finished. I will check it.
Not sure about `--skip-apps` not skipping the installation of the apps though...

lmamane, 2020-05-22:
Yes, that file contains a:
        <key>iBooks Data 2</key>
and then about 930 lines of 68 characters inside a <data> XML tag. That looks like a Bas64 encoding, so that would be 47kiB of decoded data. The decoded data seems to be binary data, that starts with "bplist", and it contains all over it what looks like ASCII-encoded UUID-like strings.

Since one of the PDFs I had in iBooks is more than 50MiB, I ran on the backup directory:

find /path/to/backup -size +50M -print0 | xargs -0 file | grep -i pdf

but none is found... This made me think it is not actually in the backup. But maybe it is there in a more hidden form?

nikias, 2020-05-22:
I know, it's the /Books/iBooksData2.plist file:
https://github.com/libimobiledevice/libimobiledevice/blob/master/tools/idevicebackup2.c#L488
Not sure where the PDFs go. I will investigate...

mexmer, 2020-05-25:
from my experiments itunes also do not restore uploaded books trough itunes, only ibooks purchases (but those are restored by itunes app inside device)

i think it's like with music, you need to sync content manually to new device.