libimobiledevice/libimobiledevice #960: Getting libimobiledevice HEAD-80eec8b did not build error

sudhir540, 2020-05-21:
Tried following commands

```
brew uninstall  libimobiledevice
brew uninstall  ideviceinstaller
[libimobiledevice_installerror.txt](https://github.com/libimobiledevice/libimobiledevice/files/4661239/libimobiledevice_installerror.txt)

brew uninstall usbmuxd
sudo rm /var/db/lockdown/*
brew install --HEAD usbmuxd
brew unlink usbmuxd
brew link usbmuxd
brew install --HEAD libimobiledevice
```

libimobiledevice is failing with below error
```

Undefined symbols for architecture x86_64:
  "_plist_string_val_compare", referenced from:
      _compproxy_get_device_registry in companion_proxy.o
      _compproxy_get_value_from_registry in companion_proxy.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [libimobiledevice.la] Error 1
make: *** [install-recursive] Error 1
```

yktan8, 2020-05-22:
I am facing same the same error too.

lcyfly, 2020-05-22:
I am facing same the same error too.

nikias, 2020-05-22:
Please update libplist. Also please pull latest libimobiledevice too. Thanks.

closed 2020-05-22 by nikias