libimobiledevice/libimobiledevice #961: macOS build libimobiledevice 1.2.1 error!

Vxer-Lee, 2020-05-22:
`libtool: compile:  clang -DHAVE_CONFIG_H -I. -I.. -I../include -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/libusbmuxd/HEAD-5364a1b/include -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/openssl@1.1/1.1.1g/include -g -O2 -c preboard.c -o preboard.o >/dev/null 2>&1
libtool: compile:  clang -DHAVE_CONFIG_H -I. -I.. -I../include -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/libusbmuxd/HEAD-5364a1b/include -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/openssl@1.1/1.1.1g/include -g -O2 -c companion_proxy.c -o companion_proxy.o >/dev/null 2>&1
libtool: compile:  clang -DHAVE_CONFIG_H -I. -I.. -I../include -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/libusbmuxd/HEAD-5364a1b/include -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/openssl@1.1/1.1.1g/include -g -O2 -c syslog_relay.c -o syslog_relay.o >/dev/null 2>&1
/bin/sh ../libtool  --tag=CC   --mode=link clang -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden -I/usr/local/Cellar/libplist/2.1.0/include -I/usr/local/Cellar/libusbmuxd/HEAD-5364a1b/include   -I/usr/local/Cellar/libplist/2.1.0/include  -I/usr/local/Cellar/openssl@1.1/1.1.1g/include  -g -O2   -L/usr/local/Cellar/libplist/2.1.0/lib -lplist -L/usr/local/Cellar/libusbmuxd/HEAD-5364a1b/lib -lusbmuxd  -L/usr/local/Cellar/openssl@1.1/1.1.1g/lib -lssl -lcrypto  -version-info 6:0:0 -no-undefined   -o libimobiledevice.la -rpath /usr/local/Cellar/libimobiledevice/HEAD-80eec8b_6/lib idevice.lo service.lo property_list_service.lo device_link_service.lo lockdown.lo afc.lo file_relay.lo notification_proxy.lo installation_proxy.lo sbservices.lo mobile_image_mounter.lo screenshotr.lo mobilesync.lo mobilebackup.lo house_arrest.lo mobilebackup2.lo misagent.lo restore.lo diagnostics_relay.lo heartbeat.lo debugserver.lo webinspector.lo mobileactivation.lo preboard.lo companion_proxy.lo syslog_relay.lo ../common/libinternalcommon.la
libtool: link: clang -dynamiclib  -o .libs/libimobiledevice.6.dylib  .libs/idevice.o .libs/service.o .libs/property_list_service.o .libs/device_link_service.o .libs/lockdown.o .libs/afc.o .libs/file_relay.o .libs/notification_proxy.o .libs/installation_proxy.o .libs/sbservices.o .libs/mobile_image_mounter.o .libs/screenshotr.o .libs/mobilesync.o .libs/mobilebackup.o .libs/house_arrest.o .libs/mobilebackup2.o .libs/misagent.o .libs/restore.o .libs/diagnostics_relay.o .libs/heartbeat.o .libs/debugserver.o .libs/webinspector.o .libs/mobileactivation.o .libs/preboard.o .libs/companion_proxy.o .libs/syslog_relay.o   -Wl,-force_load,../common/.libs/libinternalcommon.a  -L/usr/local/Cellar/libplist/2.1.0/lib -L/usr/local/Cellar/libusbmuxd/HEAD-5364a1b/lib -L/usr/local/Cellar/openssl@1.1/1.1.1g/lib -lssl -lcrypto -lusbmuxd -lplist  -g -O2   -install_name  /usr/local/Cellar/libimobiledevice/HEAD-80eec8b_6/lib/libimobiledevice.6.dylib -compatibility_version 7 -current_version 7.0 -Wl,-single_module
Undefined symbols for architecture x86_64:
  "_plist_string_val_compare", referenced from:
      _compproxy_get_device_registry in companion_proxy.o
      _compproxy_get_value_from_registry in companion_proxy.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [libimobiledevice.la] Error 1
make: *** [install-recursive] Error 1`

`==> ENV
HOMEBREW_CC: clang
HOMEBREW_CXX: clang++
MAKEFLAGS: -j8
CMAKE_PREFIX_PATH: /usr/local/opt/openssl@1.1:/usr/local/opt/readline:/usr/local/opt/sqlite:/usr/local/opt/python@3.8:/usr/local/opt/libxml2:/usr/local
CMAKE_INCLUDE_PATH: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/usr/include/apache2:/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Headers
CMAKE_LIBRARY_PATH: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Libraries
CMAKE_FRAMEWORK_PATH: /usr/local/opt/python@3.8/Frameworks:/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/System/Library/Frameworks
PKG_CONFIG_PATH: /usr/local/opt/openssl@1.1/lib/pkgconfig:/usr/local/opt/readline/lib/pkgconfig:/usr/local/opt/sqlite/lib/pkgconfig:/usr/local/opt/xz/lib/pkgconfig:/usr/local/opt/python@3.8/lib/pkgconfig:/usr/local/opt/libxml2/lib/pkgconfig:/usr/local/opt/libplist/lib/pkgconfig:/usr/local/opt/libtasn1/lib/pkgconfig:/usr/local/opt/libusb/lib/pkgconfig:/usr/local/opt/libusbmuxd/lib/pkgconfig
PKG_CONFIG_LIBDIR: /usr/lib/pkgconfig:/usr/local/Homebrew/Library/Homebrew/os/mac/pkgconfig/10.15
HOMEBREW_GIT: git
HOMEBREW_SDKROOT: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk
ACLOCAL_PATH: /usr/local/opt/libxml2/share/aclocal:/usr/local/share/aclocal
PATH: /usr/local/Homebrew/Library/Homebrew/shims/mac/super:/usr/local/opt/autoconf/bin:/usr/local/opt/automake/bin:/usr/local/opt/libtool/bin:/usr/local/opt/gdbm/bin:/usr/local/opt/openssl@1.1/bin:/usr/local/opt/sqlite/bin:/usr/local/opt/xz/bin:/usr/local/opt/python@3.8/bin:/usr/local/opt/libxml2/bin:/usr/local/opt/pkg-config/bin:/usr/local/opt/libplist/bin:/usr/local/opt/libtasn1/bin:/usr/local/opt/libusbmuxd/bin:/usr/bin:/bin:/usr/sbin:/sbin

Error: libimobiledevice HEAD-80eec8b did not build
Logs:
     /Users/lee/Library/Logs/Homebrew/libimobiledevice/02.configure
     /Users/lee/Library/Logs/Homebrew/libimobiledevice/00.options.out
     /Users/lee/Library/Logs/Homebrew/libimobiledevice/02.configure.cc
     /Users/lee/Library/Logs/Homebrew/libimobiledevice/03.make.cc
     /Users/lee/Library/Logs/Homebrew/libimobiledevice/config.log
     /Users/lee/Library/Logs/Homebrew/libimobiledevice/03.make
     /Users/lee/Library/Logs/Homebrew/libimobiledevice/01.autogen.sh
     /Users/lee/Library/Logs/Homebrew/libimobiledevice/01.autogen.sh.cc
READ THIS: https://docs.brew.sh/Troubleshooting

Please create pull requests instead of asking for help on Homebrew's GitHub,
Discourse, Twitter or IRC.`

lcyfly, 2020-05-22:
mark

nikias, 2020-05-22:
Please update libplist. Also please pull latest libimobiledevice too. Thanks.

closed 2020-05-22 by nikias

iOSleep, 2020-05-27:
@lcyfly @Vxer-Lee v
```
brew uninstall --ignore-dependencies libplist
brew install --HEAD --build-from-source libplist
```