libimobiledevice/libimobiledevice #962: idevicebackup2 stuck at requesting status.plist

mynote, 2020-05-24:
Hello,

i have a problem with idevicebackup2 during requesting a backup using java.

I've tried to copy the whole process from idevicebackup2.c but it keeps sending  `DLMessageDownloadFiles` for Status.plist.

As far as i know it always try to request the Status.plist at the beginning, but sending the following packages will lead to get `DLMessageUploadFiles` and `DLMessageGetFreeDiskSpace`:

  while (backupisRunning) {
                    PointerByReference dlFilesPointer = new PointerByReference();

                    int receivedMessage = ImobileBackup2Library.mobilebackup2_receive_message(mobileBackup2ClientHandle.getValue(), messagePointer.getPointer(), dlMessagePointer.getPointer());
                    if (receivedMessage == -4) {
                        System.out.println("Backup version mismatch..");
                        backupisRunning = false;
                        break;
                    }
                    if (messagePointer.getValue() != null) {
                        System.out.println("Got Message: " + messagePointer.getValue().getString(0));
                    }

                    if (dlMessagePointer.getValue() != null) {
                        System.out.println("Got dlMessage: " + dlMessagePointer.getValue().getString(0));

                    }

                    if (dlMessagePointer.getValue().getString(0).equals("DLMessageDownloadFiles")) {
                        mb2_set_overall_progress_from_message(messagePointer, dlMessagePointer);
                        dlFilesPointer.setValue(PlistLibrary.plist_array_get_item(messagePointer.getValue(), 3));
                        mb2_handle_send_files(mobileBackup2ClientHandle, messagePointer, infoPlistFile.getParentFile().getAbsolutePath());

                    } else if (dlMessagePointer.getValue().getString(0).equals("DLMessageUploadFiles")) {
                        mb2_set_overall_progress_from_message(messagePointer, dlMessagePointer);
                        dlFilesPointer.setValue(PlistLibrary.plist_array_get_item(messagePointer.getValue(), 3));
                        file_count += mb2_handle_receive_files(mobileBackup2ClientHandle, messagePointer, infoPlistFile.getParentFile().getAbsolutePath());

                    } else if (dlMessagePointer.getValue().getString(0).equals("DLMessageProcessMessage")) {
                        dlFilesPointer.setValue(PlistLibrary.plist_array_get_item(messagePointer.getValue(), 1));
                        if (PlistLibrary.plist_get_node_type(dlFilesPointer.getValue()) != 5) {
                            System.out.println("Received Strange node type in DLMessageProcessMessage: " + PlistLibrary.plist_get_node_type(messagePointer.getValue()));
                        } else {
                            System.out.println("NODE TYPE WAS DICT");
                        }

                        PointerByReference nn = new PointerByReference();
                        int error_code = 1;
                        nn.setValue(PlistLibrary.plist_dict_get_item(dlFilesPointer.getValue(), "ErrorCode"));

                        if (nn.getPointer() != null && PlistLibrary.plist_get_node_type(nn.getValue()) == 1) {
                            System.out.println("RECEIVED NN SUCCESSFULLY (INT). It's " + nn.getValue().getInt(0));
                            IntByReference nnError = new IntByReference();
                            PlistLibrary.plist_get_uint_val(nn.getValue(), nnError.getPointer());
                            System.out.println("NNERROR=" + nnError.getValue());
                        }

                    } else {
                        System.out.println("Received an annoying messagepointer => " + dlMessagePointer.getValue().getString(0));
                    }

nikias, 2020-05-25:
Well I can't verify all of your java code, but idevicebackup2 works so if you mimic its behavior it should work. You probably have to look closer. Try also running `idevicesyslog -p BackupAgent2` to see if the device logs any weird errors. You might need to compare the actual packages sent by running idevicebackup with `-d` and compare what gets send and what you try to send.

mynote, 2020-05-25:
Thank you for the fast answer.. i tried to debug using idevicesyslog, but i cant actually read the packages sent and received.

I just saw that an error occured several times with "No common version". 

I tried to exchange versions by using

```
double[] local_versions = {2, 2.1};

mobilebackup2_version_exchange(mobileBackup2ClientHandle.getValue(), local_versions, 2, remote_version.getPointer());
```

reworked the dll by accepting two doubles  (since it just looks through the arrays) 

`mobilebackup2_version_exchange_simple(mobileBackup2ClientHandle.getValue(), local_version_1.getPointer(), local_version_1.getPointer(), 2, remote_versions.getPointer());`

and also tried to send the request manually .. all of them failed.

   ```
      DoubleByReference local_version_1 = new DoubleByReference(2.0);
         DoubleByReference local_version_2 = new DoubleByReference(2.1);

          PointerByReference remote_versions = new PointerByReference();
          Pointer dict_version_root = PlistLibrary.plist_new_dict();
          Pointer dict_version_array = PlistLibrary.plist_new_array();
          PlistLibrary.plist_array_append_item(dict_version_array, PlistLibrary.plist_new_real(3.1));
          PlistLibrary.plist_array_append_item(dict_version_array, PlistLibrary.plist_new_real(3.2));
          PlistLibrary.plist_array_append_item(dict_version_array, PlistLibrary.plist_new_real(3.3));
          PlistLibrary.plist_dict_set_item(dict_version_root, "SupportedProtocolVersions", dict_version_array);
            int helloReturn = ImobileBackup2Library.mobilebackup2_send_message(mobileBackup2ClientHandle.getValue(), "Hello", dict_version_root);
            System.out.println("Hello returned " + helloReturn);
            PointerByReference plistReceive = new PointerByReference();
            PointerByReference dlMessage = new PointerByReference();
            int receivedMessage = ImobileBackup2Library.mobilebackup2_receive_message(mobileBackup2ClientHandle.getValue(), plistReceive.getPointer(), null);
```

Syslog shows me many of 
`May 25 18:35:54 afcd(libafc.dylib)[256] <Notice>: AFCFileLine="2594" AFCCode="-402636793" AFCFileName="server.c" AFCVersion="270" NSDescription="Invalid file ref"`
which also leads to empy received packages:

```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
</plist>

```

this is the error i get at the beginning:

```
May 25 18:38:07 BackupAgent2[426] <Notice>: Device UDID: <private>
May 25 18:38:07 BackupAgent2(liblockdown.dylib)[426] <Notice>: unlock_with_escrow: <private>
May 25 18:38:08 BackupAgent2(MobileKeyBag)[426] <Notice>: MKBKeyBagUnlock: unlocked bag 1
May 25 18:38:08 BackupAgent2[426] <Error>: No common supported protocol version
May 25 18:38:09 BackupAgent2(MobileDeviceLink)[426] <Notice>: 426:1809461248|SocketStreamHandler.c:_SocketLogCallback| ERROR: SocketRecv (data): error -555: Unknown error: -555
May 25 18:38:09 BackupAgent2(MobileDeviceLink)[426] <Notice>: 426:1809461248|StreamHandler.c:_ReadFunc| ERROR: Error reading data: Can't read data
May 25 18:38:09 BackupAgent2(MobileDeviceLink)[426] <Notice>: 426:1809461248|DeviceLinkConnection.c:_DLHandlerThreadReadMessage| ERROR: Could not receive message: '(null)'
May 25 18:38:09 BackupAgent2(MobileDeviceLink)[426] <Notice>: 426:73357440|DeviceLinkConnection.c:_DLMainThreadMessagePortCallback| ERROR: Error creating components from read data: 'NULL data returned from read'
```

nikias, 2020-05-25:
Yes you got some serious issues with the library. I can't help you with java though since this is out of scope of this project.

mynote, 2020-05-25:
thank you again for the fast answer .. could you proivde a simple workflow which packages need to be sent to get a files list?

im able to get the request for Status.plist and sending the error back .. but after that, i only get empy plist packages. Which versions do i need to send? I saw somewhere else that he sent 3.0, 3.1, 3.2 (instead of 2.0, 2.1).