libimobiledevice/libimobiledevice #970: SSL errors when backing up

dfoody, 2020-06-08:
80eec8b83cca16d7a7c7cf9fc53422505cf31225 introduced support for non-blocking SSL, but it isn't handling EAGAIN or ETIMEDOUT errors.

`socket_check_fd` returns `-ETIMEDOUT` when it times out, but this change appears to expect it to return 0. So, a timeout results in `idevice_connection_send` returning an SSL error. Also, when `socket_check_fd` experiences an `EAGAIN` error, this also leads to an error being returned by `idevice_connection_send` (I'm not sure if the right solution is to change `socket_check_fd` to set `eagain = 1` on `EAGAIN` or to handle the eagain error with a `continue` in `idevice_connection_send`).

nikias, 2020-06-08:
Check commit 960fb407eb362bb0e90326a30a984c1af1e5b0dd.

nikias, 2020-06-08:
To be fair, I tried it with blocking or non-blocking socket (via libusbmuxd) and I did not trigger any issue. Nevertheless I added a specific check for it in the commit that should handle it if it occurs.

closed 2020-06-08 by nikias

dfoody, 2020-06-08:
I did not experience the issue when backing up over USB - but when I was backing up over WiFi I could fairly reliably reproduce it - 3/4 of backups would hang when sending manifest.db over WiFi because of this error (it's a little tricky to spot without lower level debugging on - because `mb2_handle_send_files` silently ignores errors that occur when sending files).

nikias, 2020-06-08:
Ah ok. I assume this solves it?

dfoody, 2020-06-09:
@nikias yep, all good. Thx.