libimobiledevice/libimobiledevice #971: could not connect to lockdownd error code -3

jhachandan1994, 2020-06-08:
On all mac system with version 10.15.5 i am getting error as could not connect to lockdownd error code -3 while using command idevicescreenshot / ideviceinstaller / ideviceinfo i have tried reinstalling libimobiledevice / reconnecting ios devices etc. but i am unable to use any commands on any laptop with 10.15.5 version of mac

aomegax, 2020-06-15:
me too on Catalina 10.15.4

but it is a problem about libimobiledevice/1.2.0_6 because on another mac having libimobiledevice/HEAD-3d8d13f_6 it works properly.

do you know how to install a specific version using brew?

emcas88, 2020-06-16:
I am experiencing the same issue in catalina version 10.15.5 and libimobiledevice/1.2.0_6