libimobiledevice/libimobiledevice #972: ERROR: Package requirements (libusbmuxd-2.0 >= 2.0.2) were not met

devlucy, 2020-06-09:
```
checking for libusbmuxd-2.0 >= 2.0.2... no
configure: error: Package requirements (libusbmuxd-2.0 >= 2.0.2) were not met:

No package 'libusbmuxd-2.0' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libusbmuxd_CFLAGS
and libusbmuxd_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
```

nikias, 2020-06-10:
We are about to release new versions of all the libs and tools. This error is because you need a newer libusbmuxd: https://github.com/libimobiledevice/libusbmuxd, and you will also need newer libplist: https://github.com/libimobiledevice/libplist

closed 2020-06-10 by nikias

xyd113, 2020-06-12:
Same issue.

hokiepokedad2, 2020-06-15:
Any update to this? Having the same issues as original issue.

nikias, 2020-06-15:
Hi, please check out the latest sources for all dependencies (libplist, libusbmuxd) from git and build them before trying to build latest libimobiledevice.

hokiepokedad2, 2020-06-15:
Ah okay. I was trying to install ideviceinstaller and was getting the same. Does that need the latest as well?

workhardupc, 2020-06-16:
> Ah okay. I was trying to install ideviceinstaller and was getting the same. Does that need the latest as well?

first clone libplist、libusbmuxd:

https://github.com/libimobiledevice/libplist
https://github.com/libimobiledevice/libusbmuxd.git

sencond execute: ./autogen.sh   、 make  、make install

greyreality, 2020-06-16:
> > Ah okay. I was trying to install ideviceinstaller and was getting the same. Does that need the latest as well?
> 
> first clone libplist、libusbmuxd:
> 
> https://github.com/libimobiledevice/libplist
> https://github.com/libimobiledevice/libusbmuxd.git
> 
> sencond execute: ./autogen.sh 、 make 、make install

Hello @workhardupc 
i did 
`git clone https://github.com/libimobiledevice/libplist.git
cd libplist
./autogen.sh
`

this is the log
`glibtoolize: putting auxiliary files in '.'.
glibtoolize: linking file './ltmain.sh'
glibtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
glibtoolize: linking file 'm4/libtool.m4'
glibtoolize: linking file 'm4/ltoptions.m4'
glibtoolize: linking file 'm4/ltsugar.m4'
glibtoolize: linking file 'm4/ltversion.m4'
glibtoolize: linking file 'm4/lt~obsolete.m4'
configure.ac:29: installing './compile'
configure.ac:6: installing './missing'
cython/Makefile.am: installing './depcomp'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... ./install-sh -c -d
checking for gawk... no
checking for mawk... no
checking for nawk... no
checking for awk... awk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... clang
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether clang accepts -g... yes
checking for clang option to accept ISO C89... none needed
checking whether clang understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of clang... gcc3
checking whether we are using the GNU C++ compiler... yes
checking whether clang++ accepts -g... yes
checking dependency style of clang++... gcc3
checking whether clang++ is available and compiles a program... yes
checking build system type... x86_64-apple-darwin17.7.0
checking host system type... x86_64-apple-darwin17.7.0
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/local/bin/gsed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by clang... /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld
checking if the linker (/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld) is GNU ld... no
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 196608
checking how to convert x86_64-apple-darwin17.7.0 file names to x86_64-apple-darwin17.7.0 format... func_convert_file_noop
checking how to convert x86_64-apple-darwin17.7.0 file names to toolchain format... func_convert_file_noop
checking for /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld option to reload object files... -r
[... 137 lines cut ...]
and i tried again
`brew install --HEAD libimobiledevice`
still get an error

> No package 'libusbmuxd-2.0' found
> Consider adjusting the PKG_CONFIG_PATH environment variable if you
> installed software in a non-standard prefix.
> Alternatively, you may set the environment variables [...]

`brew doctor`
said
> Warning: Unbrewed .la files were found in /usr/local/lib.
> If you didn't put them there on purpose they could cause problems when
> building Homebrew formulae, and may need to be deleted.
> Unexpected .la files:
>   /usr/local/lib/libplist++-2.0.la
>   /usr/local/lib/libplist-2.0.la
>   Unexpected .pc files:
>     /usr/local/lib/pkgconfig/libplist++-2.0.pc
>     /usr/local/lib/pkgconfig/libplist-2.0.pc
>   Unexpected static libraries:
>     /usr/local/lib/libplist++-2.0.a
>     /usr/local/lib/libplist-2.0.a

Have no idea what to do xD I just wanted to install ipa from macbook onto iphone(

workhardupc, 2020-06-17:
> > > Ah okay. I was trying to install ideviceinstaller and was getting the same. Does that need the latest as well?
> > 
> > 
> > first clone libplist、libusbmuxd:
> > https://github.com/libimobiledevice/libplist
> > https://github.com/libimobiledevice/libusbmuxd.git
> > sencond execute: ./autogen.sh 、 make 、make install
> 
> Hello @workhardupc
> i did
> 
> > git clone https://github.com/libimobiledevice/libplist.git
> > cd libplist
> > ./autogen.sh
> 
> this is the log
> 
> ```
> glibtoolize: linking file './ltmain.sh'
> glibtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
> glibtoolize: linking file 'm4/libtool.m4'
> glibtoolize: linking file 'm4/ltoptions.m4'
> glibtoolize: linking file 'm4/ltsugar.m4'
> glibtoolize: linking file 'm4/ltversion.m4'
> glibtoolize: linking file 'm4/lt~obsolete.m4'
> configure.ac:29: installing './compile'
> configure.ac:6: installing './missing'
> cython/Makefile.am: installing './depcomp'
> checking for a BSD-compatible install... /usr/bin/install -c
> checking whether build environment is sane... yes
> checking for a thread-safe mkdir -p... ./install-sh -c -d
> checking for gawk... no
> checking for mawk... no
> checking for nawk... no
> checking for awk... awk
> checking whether make sets $(MAKE)... yes
> checking whether make supports nested variables... yes
> checking whether make supports nested variables... (cached) yes
> checking for gcc... clang
> checking whether the C compiler works... yes
> checking for C compiler default output file name... a.out
> checking for suffix of executables...
> checking whether we are cross compiling... no
> checking for suffix of object files... o
> checking whether we are using the GNU C compiler... yes
> checking whether clang accepts -g... yes
> checking for clang option to accept ISO C89... none needed
> checking whether clang understands -c and -o together... yes
> checking whether make supports the include directive... yes (GNU style)
> checking dependency style of clang... gcc3
> checking whether we are using the GNU C++ compiler... yes
> checking whether clang++ accepts -g... yes
> checking dependency style of clang++... gcc3
> checking whether clang++ is available and compiles a program... yes
> checking build system type... x86_64-apple-darwin17.7.0
> checking host system type... x86_64-apple-darwin17.7.0
> checking how to print strings... printf
> checking for a sed that does not truncate output... /usr/local/bin/gsed
> checking for grep that handles long lines and -e... /usr/bin/grep
> checking for egrep... /usr/bin/grep -E
> checking for fgrep... /usr/bin/grep -F
> checking for ld used by clang... /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld
> checking if the linker (/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld) is GNU ld... no
> checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
> checking the name lister (/usr/bin/nm -B) interface... BSD nm
> checking whether ln -s works... yes
> checking the maximum length of command line arguments... 196608
> checking how to convert x86_64-apple-darwin17.7.0 file names to x86_64-apple-darwin17.7.0 format... func_convert_file_noop
> checking how to convert x86_64-apple-darwin17.7.0 file names to toolchain format... func_convert_file_noop
> checking for /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld option to reload object files... -r
[... 146 lines cut ...]
> > Alternatively, you may set the environment variables libusbmuxd_CFLAGS
> > and libusbmuxd_LIBS to avoid the need to call pkg-config.
> > See the pkg-config man page for more details.
> 
> `brew doctor`
> said
> 
> > Warning: Unbrewed .la files were found in /usr/local/lib.
> > If you didn't put them there on purpose they could cause problems when
> > building Homebrew formulae, and may need to be deleted.
> > Unexpected .la files:
> > /usr/local/lib/libplist++-2.0.la
> > /usr/local/lib/libplist-2.0.la
> > Unexpected .pc files:
> > /usr/local/lib/pkgconfig/libplist++-2.0.pc
> > /usr/local/lib/pkgconfig/libplist-2.0.pc
> > Unexpected static libraries:
> > /usr/local/lib/libplist++-2.0.a
> > /usr/local/lib/libplist-2.0.a
> 
> Have no idea what to do xD I just wanted to install ipa from macbook onto iphone(

fisrt git clone https://github.com/libimobiledevice/libimobiledevice.git
then ./autogen.sh 、 make 、make install
don't use this "brew" install libimobiledevice

rakurati7, 2020-06-17:
@workhardupc QQ, from your last two comments  you said

>First comment

first clone libplist、libusbmuxd:
https://github.com/libimobiledevice/libplist
https://github.com/libimobiledevice/libusbmuxd.git

sencond execute: ./autogen.sh 、 make 、make install

>Second Comment

fisrt git clone https://github.com/libimobiledevice/libimobiledevice.git
then ./autogen.sh 、 make 、make install
don't use this "brew" install libimobiledevice

My question is what is the order we have to follow here what needs to done first? libplist, libusbmuxd needs to be installed first and then install libimobiledevice.git

rakurati7, 2020-06-18:
Below steps worked for me 
1. brew update 
2. brew uninstall --ignore-dependencies libimobiledevice 
3. brew uninstall --ignore-dependencies usbmuxd 
4. brew install --HEAD usbmuxd
5. brew unlink usbmuxd (or) bew unlink libusbmuxd
6. brew link usbmuxd (or) brew link libusbmuxd (if there is any linking issue then use brew link --over-write libusbmuxd)
7. brew install --HEAD libimobiledevice 
8. brew install ideviceinstaller

After this when I ran ideviceinfo it works fine with out any issue

CoolBoyHwq, 2020-12-07:
@rakurati7 
My centos system does not solve the problem according to the operation

[root@localhost libimobiledevice]# git  clone  https://github.com/libimobiledevice/libplist
Cloning into 'libplist'...
remote: Enumerating objects: 4393, done.
remote: Total 4393 (delta 0), reused 0 (delta 0), pack-reused 4393
Receiving objects: 100% (4393/4393), 1.31 MiB | 992.00 KiB/s, done.
Resolving deltas: 100% (2719/2719), done.
[root@localhost libimobiledevice]# git  clone  https://github.com/libimobiledevice/libusbmuxd.git
Cloning into 'libusbmuxd'...
remote: Enumerating objects: 10, done.
remote: Counting objects: 100% (10/10), done.
remote: Compressing objects: 100% (10/10), done.
remote: Total 792 (delta 0), reused 2 (delta 0), pack-reused 782
Receiving objects: 100% (792/792), 268.01 KiB | 294.00 KiB/s, done.
Resolving deltas: 100% (447/447), done.
[root@localhost libimobiledevice]# ./autogen.sh
libtoolize: putting auxiliary files in `.'.
libtoolize: linking file `./ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
libtoolize: linking file `m4/libtool.m4'
libtoolize: linking file `m4/ltoptions.m4'
libtoolize: linking file `m4/ltsugar.m4'
libtoolize: linking file `m4/ltversion.m4'
libtoolize: linking file `m4/lt~obsolete.m4'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking whether gcc and cc understand -c and -o together... yes
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking how to convert x86_64-unknown-linux-gnu file names to x86_64-unknown-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-unknown-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
[... 36 lines cut ...]
checking how to run the C++ preprocessor... g++ -E
checking for ld used by g++... /usr/bin/ld -m elf_x86_64
checking if the linker (/usr/bin/ld -m elf_x86_64) is GNU ld... yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking for g++ option to produce PIC... -fPIC -DPIC
checking if g++ PIC flag -fPIC -DPIC works... yes
checking if g++ static flag -static works... no
checking if g++ supports -c -o file.o... yes
checking if g++ supports -c -o file.o... (cached) yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for libusbmuxd... no
configure: error: Package requirements (libusbmuxd-2.0 >= 2.0.2) were not met:

No package 'libusbmuxd-2.0' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libusbmuxd_CFLAGS
and libusbmuxd_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

nikias, 2020-12-07:
@CoolBoyHwq you clone libplist and libusbmux from git but don't build and install them and then wonder that libimobiledevice's configure doesn't find them? Just build and install all dependencies and it will work.

CoolBoyHwq, 2020-12-07:
@nikias Sorry, first of all, I am a novice. After entering libplist and executing ./autogen.sh, make, and make install, it is successful, but after entering libusbmux and executing ./autogen.sh, it still reports that error. Is it my way?

[root@localhost libusbmuxd]# ls
aclocal.m4  AUTHORS  autogen.sh  autom4te.cache  common  config.h.in  config.log  configure  configure.ac  COPYING  docs  include  m4  Makefile.am  Makefile.in  NEWS  README.md  src  tools
[root@localhost libusbmuxd]# ./autogen.sh 
libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
libtoolize: linking file `m4/libtool.m4'
libtoolize: linking file `m4/ltoptions.m4'
libtoolize: linking file `m4/ltsugar.m4'
libtoolize: linking file `m4/ltversion.m4'
libtoolize: linking file `m4/lt~obsolete.m4'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking dependency style of gcc... gcc3
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking dependency style of gcc... (cached) gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking whether gcc and cc understand -c and -o together... yes
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking how to convert x86_64-unknown-linux-gnu file names to x86_64-unknown-linux-gnu format... func_convert_file_noop
[... 30 lines cut ...]
checking how to run the C++ preprocessor... g++ -E
checking for ld used by g++... /usr/bin/ld -m elf_x86_64
checking if the linker (/usr/bin/ld -m elf_x86_64) is GNU ld... yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking for g++ option to produce PIC... -fPIC -DPIC
checking if g++ PIC flag -fPIC -DPIC works... yes
checking if g++ static flag -static works... no
checking if g++ supports -c -o file.o... yes
checking if g++ supports -c -o file.o... (cached) yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for libplist... no
configure: error: Package requirements (libplist-2.0 >= 2.2.0) were not met:

No package 'libplist-2.0' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libplist_CFLAGS
and libplist_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

nikias, 2020-12-07:
As suggested by the configure output, try this:
```
PKG_CONFIG_PATH=/usr/local/lib/pkgconfig ./autogen.sh
```

CoolBoyHwq, 2020-12-07:
@nikias Solved it, thanks