libimobiledevice/libimobiledevice #973: diagnostics_relay_client_t issues restarting device on port.

Adderally, 2020-06-11:
`idevice_t                       device;`
`lockdownd_client_t              client;`
`lockdownd_service_descriptor_t  lock_down_service;`
`diagnostics_relay_client_t      dclient;`

`idevice_new(&device, NULL);`
`lockdownd_client_new(device, &client, "IDK WHAT THIS MEANS");`
`lockdownd_start_service(client, "IDK WHAT THIS MEANS", &lock_down_service);`
`diagnostics_relay_client_new(device, lock_down_service, &dclient);`

`diagnostics_relay_restart(dclient, DIAGNOSTICS_RELAY_ACTION_FLAG_WAIT_FOR_DISCONNECT);`
`diagnostics_relay_client_free(dclient);`

Error in **lockdown.c** lines 1281 - 1285
`if (*service)`
*// reset fields if service descriptor is reused*
`(*service)->port = 0;`
`(*service)->ssl_enabled = 0;`

FunkyM, 2020-06-15:
Great, you are almost there.

However your code needs some fixes:
* Initialize `idevice_t`, `lockdownd_client_t`, `lockdownd_service_descriptor_t` and `diagnostics_relay_client_t` to `NULL`
* Add logic to handle error codes to only proceed on `*_E_SUCCESS` results
* Use `lockdownd_client_new_with_handshake()` and check that the label argument is just an identifier of your app from the [documentation](https://libimobiledevice.org/docs/html/lockdown_8h.html#func-members).
* Use the `DIAGNOSTICS_RELAY_SERVICE_NAME` define for the `lockdownd_start_service()` identifier argument (the one you labeled with `IDK WHAT THIS MEANS`)
* Add a call to `diagnostics_relay_goodbye()` before calling `diagnostics_relay_client_free()`

This should get your code running.

closed 2020-06-15 by FunkyM

Adderally, 2020-06-15:
It worked! thank you so much!

Next thing i need to tackle is _**Factory data resetting**_

Wonder if initializing them to null will help? cause I've gotten that same error mentioned above on almost everything till now (setting them to NULL)