libimobiledevice/libimobiledevice #977: How i can dump IOKitDiagnostics classes data using idevicediagnostics

Johnny314, 2020-06-18:
i am trying to dump IOKitDiagnositcs classes which is in following format 
+-o Root  <class IORegistryEntry, id 0x100000100, retain 35>
  | {
  |   "IOKitBuildVersion" = "Darwin Kernel Version 19.3.0: Thu Jan  9 21:10:55 PST 2020; root:xnu-6153.82.3~1/RELEASE_ARM64_T8010"
  |   "OS Build Version" = "17D50"
  |   "OSKernelCPUSubtype" = 0
  |   "OSKernelCPUType" = 16777228
  |   "OSPrelinkKextCount" = 215
  |   "IORegistryPlanes" = {"IO80211Plane"="IO80211Plane","IOPower"="IOPower","IODeviceTree"="IODeviceTree","IOService"="IOService","IOAccessory"="IOAccessory","CoreCapture"="CoreCapture"}
  |   "IOConsoleLocked" = Yes
  |   "IOKitDiagnostics" = {"Instance allocation"=8553904,"Container allocation"=3256314,"Pageable allocation"=46743552,"Classes"={"NeighborCache"=1,"IOBioSEPSharedBufferFactory"=0,"AppleARMFunctionChargerMux"=0,"AppleS5L8920XPWM"=1}

Johnny314, 2020-06-18:
@FunkyM  please take a look

msk521, 2021-12-06:
but when I use idevicediagnostics ioregentry IOAccessory ,returned empty