libimobiledevice/libimobiledevice #979: restore.h can't connect to device UDID in recovery.

Adderally, 2020-06-20:
I know i've posted a bunch so far, but every time i do someone ends up helping out lol... so the last 2 things i want to do are

1. Factory data reset a device.
2. Flash a device with a ipsw file.

`idevice_t                           device = NULL;`
`restored_client_t                   client = NULL;`

`idevice_new(&device, NULL);`
`restored_client_new(device, &client, "NAME");`

`restored_reboot(client);`

when the device is in recovery i get error raised at _line 309_ `debug_info("could not connect to restored (device %s)", device->udid);` in file **restore.c**

I also get the same error when trying `restored_start_restore(client, NULL, NULL);` .. device also in recovery mode.

kalit-zz, 2021-09-29:
HI William,

We are facing this kind of issue for new iOS 15, we are unable to get out of recovery mode.
Are you facing this kind of issue for iOS 15 or not?

nikias, 2021-09-29:
@WilliamHottzz The above issue doesn't make sense. The device is in RECOVERY mode so you cannot connect to restored since that requires RESTORE mode. Check how ideviceinfo works.
@kalit-zz you keep posting issues without stating what you are trying to do. Provide commands you run, output you get, etc.
Entering and leaving recovery works just fine.

closed 2021-09-29 by nikias

WilliamHottzz, 2021-10-03:
@nikias is there any form of a factory reset? I've seen 3utools do it without firmware being installed... I've plugged it into IDA and seen a bunch of calls to idevicebackup2, any idea how that would be possible?