libimobiledevice/libimobiledevice #982: Need guidance on building libimobiledevice on Windows

kakyoism, 2020-06-24:
Up till version 1.2.0, I was successful in building libimobiledevice/libusbmuxd/libplist on Windows using VS2017. It needed a bit of tweaking such as explicit typecast and proper dllexport tags. Things work fine after a successful build, e.g., I can safely discover and talk with iOS devices over USB.

But with 1.3.0, the program start to fail after fixing hundreds of tedious build errors.  Now with the following combination

- libimobiledevice 1.3.0
- libusbmuxd 2.02
- libplist 2.2.0

I get a crash in calling `idevice_get_device_list`

[image]

Could this be the wrong combination of the versions or simply a hacking problem?

What's the canonical way of building libimobiledevice on Windows? 
Should I use msys2 / mingw? The README.md is not super clear about this.

Please  help!

kakyoism, 2020-06-28:
The crash is gone after I downgrade `libplist` from `2.2.0` to `2.1.0`.

kakyoism, 2020-06-28:
But it would still be great to have a comprehensive guide for building on Windows.

Currently a lot of manual fixing has to be done for every new release to be able to build correctly with Visual Studio 2017, to name a few:

- `winsock.h` and `winsock2.h` order problem
- Some Unix headers have not been properly isolated from Windows headers, sometimes even without conditional includes for Windows 
- Lack of explicit typecasts that VS demands
- dllexport tags were not added to most API declarations
- Some headers such as thread.h lack extern "C" for C++ code.

closed 2020-06-28 by kakyoism

winstondu, 2020-08-17:
@kakyoism , can you provide a full comprehensive guide on how to build this project on windows for dummies?

kakyoism, 2020-08-18:
> @kakyoism , can you provide a full comprehensive guide on how to build this project on windows for dummies?

I recommend you to try MS package manager `vcpkg` first if you only need the binaries.
If you absolutely need to build from source, you could start with [this project](https://github.com/libimobiledevice-win32/libimobiledevice-vs).