libimobiledevice/libimobiledevice #986: Debug flag doesn't produce debug output

TheProgramableTurtle, 2020-07-04:
I've built libplist, usbmuxd, libusbmuxd, and libimobiledevice with
```
./autogen.sh --enable-debug-code
make
make install
```
but when running `ideviceinfo -d` there is no debug output, only `ERROR: No device found!`. In addition, when running `idevicepair`, I get this
```
-bash: /usr/bin/idevicepair: No such file or directory
```
Neither libimobiledevice-utils or usbmuxd are installed, so why isn't there a debug output?

TheProgramableTurtle, 2020-07-04:
Ok, not changed in the README but it’s now `—enable-debug`, will try in the morning with a fresh dev env.

TheProgramableTurtle, 2020-07-05:
Still doesn't produce a debug output, going to use something else.

closed 2020-07-05 by TheProgramableTurtle

TheNegativeZero, 2020-07-05:
@TheProgramableTurtle  : This looks very similar to a problem I am facing with idevicerestore.  It seems to me that the underlying problem is that there is no (stable) USB communication with the device.

Open a second terminal window and run "usbmuxd -f -vv" and check the messages while running your program.  You might find that either (1) the host never connects with the device, or (2) the device disconnects/times out after a few actions.

So my take on this is that the troubleshooting has to be on the usbmuxd / libusbmuxd side, and libimobiledevice.  In my case, "idevicerestore -l -d" shows additional debug information since I compiled libimobiledevice with "--enable-debug".

Also check if all required dependencies and libraries are up-to-date and properly linked.

You can also modify the C programs to add your own tracking and debugging and recompile.

Hope this helps.

TheProgramableTurtle, 2020-07-16:
@TheNegativeZero Thanks for the reply, I was trying to diagnose the connection (the device showed as connected but was not detected by any software). In the end I gave up and used another computer, but thanks for the advice regardless, hopefully it helps someone else.

jkcoxson, 2022-05-02:
This is still an issue, cannot debug connections without output.

nikias, 2022-05-03:
What exactly does not work now? If compiled with `--enable-debug` it will definitely print debug information. I am using it all the time.