libimobiledevice/libimobiledevice #987: How can I list device_id via wifi?

sbilly, 2020-07-05:
Hi

I want to use linux and macos to manage my iphone.

It works fine in my macOS laptop with wifi:

```bash
idevice_id -l -n
4c9b xxxx 14aa (Network)
```
and works fine when I connect it to linux with USB cable:

```bash
idevice_id -l -n
4c9b xxxx 14aa (USB)
```

but not working when I connect it to linux with wifi. It returns nothing.
```bash
idevice_id -l -n
```
How can I make it work in linux ? thx.

valinet, 2021-01-19:
Have you found the answer? Is there no one in this entire world that uses this library like that, I can't believe it. For what is worth, I think a lot of commercial products use this library as the backend, so I don't understand, why waste such a great effort with such a so lackluster support and documentation...

mexmer, 2021-01-19:
on linux there is no service, that is waiting or scaning for iOS devices on network to provide their discovery, that's why it's not working.

valinet, 2021-01-19:
Okay, so, how do we make it work then? Is there any existing code that we can link together? Can usbmuxd connect to Apple Mobile Device Support running on some other Windows box? What's a good starting point to have this feature implemented?

mexmer, 2021-01-22:
i don't know about any implementation on linux. and no, you cannot connect to windows machine from this.

valinet, 2021-01-22:
Found the solution: someone implemented a usbmux that is a drop in replacement for libimobiledevice/usbmux with WiFi support: https://github.com/tihmstar/usbmuxd2. Awesome, really amazing work, now Linux has this feature too.