libimobiledevice/libimobiledevice #988: Unable to read or view past mailing list messages

noloader, 2020-07-10:
I'm trying to browse past mailing list messages. The mailing list is at [List Info | libimobiledevice-devel](https://lists.libimobiledevice.org/mailman/listinfo/libimobiledevice-devel). Clicking on **_Archives_** results in:

[image]

I think this is a **_really_** bad idea. It wastes my time because I can't find information previously discussed on the mailing list. Then, it wastes the project's time because I have to ask a question that's already been asked.

I also think the hidden archives runs counter to GNU's philosophies. The project appears to be aligned with GNU for the source code, but then refuses to share some information about the source code. I think its a big contradiction.

This practice may help explain why the project has so much mailing list type questions in the bug tracker. People cannot find information they need because the information is hidden from them and not indexed by search engines.

My apologies if I am missing something obvious here.

DCKcode, 2021-01-19:
I've just subscribed to this mailing list, but it doesn't seem active at all. It looks to me that the maintainers prefer GitHub issues like this?

valinet, 2021-01-19:
I don't get it either, the documentation is almost non-existent. As mentioned above, I am looking on how to enable the device to be detected via WiFi on Linux. Does anyone know the answer? I'd appreciate it if someone could share it here, without any further unnecessary abstraction (I subscribed to the mailing list, I did not understand exactly what I am supposed to do further; when I login with the credentials on the page above I get rejected).

DCKcode, 2021-01-19:
@valinet You're right, this is all not documented. I was also on a quest to get WiFi support on Linux (for backups). This is what I found:

- `libimobiledevice` and `idevicebackup2` support wifi backups. However, they require support in `usbmuxd` for this to work. `usbmuxd` is the component that actually connects your computer to your iDevice. Apple supplies their own proprietary `usbmuxd` with macOS that supports wifi connections, and you can use `idevicebackup2` on Macs with that to support it.
- However, the open source implementation of `usbmuxd` provided by this project does not support connecting to devices over Wifi yet - I think.
- The good news: somebody else made a fork of `usbmuxd` and `libimobiledevice` to support wifi backups on Linux: https://github.com/tihmstar/usbmuxd2 . To get this to build, you also need to build and install this project's `libplist`, and subsequently build and install this other person's fork of [`libimobiledevice`](https://github.com/tihmstar/libimobiledevice) and their own [`libgeneral`](https://github.com/tihmstar/libgeneral) as well.

With the forked `usbmuxd2` running (and no system provided `usbmuxd`s) and the forked `libimobiledevice` tools, after USB pairing your device you can toggle the wifi connection by issuing `idevicepair wifi on`. When you disconnect your device, it should still show with `idevice_id`, and you can e.g. start a backup using `idevicebackup2 backup --network /some/directory`.

valinet, 2021-01-19:
@DCKcode Thank you for the reply. It can save so much time for people. Like, I just now have read it, and in the meantime since my original post, I was just able to find out about usbmuxd2 by sheer luck via Google and also understood the architecture regarding usbmuxd. So, 2 hours for that. Maybe I would have figured the rest out as well, in like 1 day or so. You just spared me that, which is incredible, and I really appreciate that, especially considering you have probably wasted your own day some time ago to figure all that out. That's why I also think a project's value is severely diminished, especially when published in a place like GitHub, unless good documentation accompanies it.

At the moment, compilation of usbmuxd2 failed, probably because my gcc/g++ 8 is too old (undefined reference to `_atomic_load_8') - it says on the project page that one needs C++17 support. Also, I will try to port the changes in the forked libimobiledevice to the official, they seem to be minor, so maybe a patch would have been more feasible there.

Again, thank you very much for your information, maybe I will put together a guide in the end, it seems it is worth it.

DCKcode, 2021-01-19:
No problem. Just went down this rabbit hole yesterday. I was able to build
all of this in a Ubuntu 20.04 container if that helps you.

I’m grateful to the authors for making all of this. It seems that wifi
backup is a pretty new thing for this project, so it’s somewhat
understandable this is underdocumented. Documenting this as part of
‘community documentation’ is fine, although better visibility of what you
can and can’t do with this project would go a long way.

valinet, 2021-01-20:
@DCKcode Thank you again, I have managed to set something up for my use case. I am doing it all on a Raspberry Pi, running Raspbian Buster, there were a few minor tweaks to be done here and there, but nothing major. Indeed, I have published a small write-up on my blog with some step-by-step instructions on how to set it up for a use case similar to mine: https://valinet.ro/2021/01/20/Automatically-backup-the-iPhone-to-the-Raspberry-Pi.html

Also, as you did, I have to thank the developers as well for all the hard work they have put into this, these are truly amazing projects.

DCKcode, 2021-01-20:
Ah that is great! Glad to see you've got it working and that you've documented it so extensively.

valinet, 2021-01-20:
Thanks. Now I realize, maybe I should have added a link to this issue as well, and also a mention for your help. Yeah, it simply slipped off my mind, I will do it properly tomorrow.