libimobiledevice/libimobiledevice #989: iPad is not recognised when in recovery mode

holesome, 2020-07-12:
Whenever I put my iPad mini (model A1432) in recovery mode by using the `ideviceenterrecovery` command from this tool or by using the `idevicerestore --erase --latest` command of [idevicerestore](https://github.com/libimobiledevice/idevicerestore), the connection to the iPad gets lost and it can't be detected anymore. The iPad is still detectable by `lsusb` but I cannot use any of the libimobiledevice commands to communicate with it. I have tried this on Arch, Kali and Debian and it gave the same result.

usbmuxd is correctly installed, as far as I know, but it doesn't start when I plug the iPad in the PC. I have to manually start it.

Does anyone know how I can solve this issue? I would like to perform a firmware restore with the [idevicerestore](https://github.com/libimobiledevice/idevicerestore) utility, but part of this process is putting the iPad in recovery mode and whenever that happens, it cannot be detected anymore and the program (idevicerestore) stops.

Let me know if I need to provide further details. Thanks in advance for any kind of help.

nikias, 2020-07-12:
Devices in recovery mode don't use usbmux. The library libirecovery is responsible for the communication, and it comes with a udev rule that configures the usb device node so the normal user has access. If correctly installed it should "just work" but apparently it is not in your case.
A quick while not recommended workaround is running isevicerestore as root, while the real fix would be to check if the udev rule is correctly installed and the user is in the `plugdev` group (or adapting the udev rule to use a different group the normal user is a member of).

sk-david, 2020-07-12:
I am experiencing the same issue. Running as root doesn't change anything for me. The tools are still unable to see the Apple device.