libimobiledevice/libimobiledevice #990: Failed to start Webdriveragent - Appium

balaganes, 2020-07-13:
I'm trying to start Appium's webdriveragent in real device using  **idevicedebug -d run com.facebook.WebDriverAgentRunner.xctrunner** I got the below error in device log. It seems like trying to contact back with XCode IDE.

`default	20:20:44.131175+0530	testmanagerd	Not permitting UI test control for client with pid 2281, no matching control session found`

I've checked the device logs when test via xcode, there are some logs related to attaching the IDE with test sessions.
Is it possible to add IDE to test session before starting the actual xctest application using libimobiledevice.

```
default	13:57:52.849513+0530	testmanagerd	IDE connected via lockdown over socket 4, with SSL
default	13:57:52.849628+0530	testmanagerd	Created session with socket 4
default	13:57:52.850819+0530	testmanagerd	Adding <XCIDESession: 0x13f21c090> (socket 4) created Monday, 13 July 2020 at 1:57:52 PM India Standard Time to the set of unidentified IDE sessions
default	13:57:52.851583+0530	testmanagerd	Handled a proxy request from the IDE with session <XCIDESession: 0x13f21c090> (socket 4) created Monday, 13 July 2020 at 1:57:52 PM India Standard Time proxy <_DTXProxy: 0x13dd21180>
```

Please help me to sort out the issue.

vikram-chaitanya, 2021-02-11:
I am also facing the same. Any luck with this?

balaganes, 2021-02-11:
No Luck 👎 
Got below message from one of the Contributors of Libimobile device.

`We do not have a full public implementation that is able to do that. One would have to glue bits and pieces manually now to do this. Furthermore, your question affects services that only work with a mounted developer disk image and are not really part of the default iOS services we primarily target for the API reimplementation for now.`