libimobiledevice/libimobiledevice #994: WiFi diagnostics - always not active

tomaszoraczewski-silvair, 2020-07-28:
`idevicediagnostics diagnostics WiFi` regardless of actual WiFi state on device returns:
```<key>WiFi</key>
	<dict>
		<key>Active</key>
		<string>NO</string>
		<key>Status</key>
		<string>Success</string>
	</dict>
```
Device info:
```ActivationState: Activated
ActivationStateAcknowledged: true
BasebandStatus: NoTelephonyCapabilty
BluetoothAddress: f0:76:6f:79:4f:e0
BoardId: 16
BrickState: false
BuildVersion: 17F80
CPUArchitecture: arm64
ChipID: 32771
DeviceClass: iPad
DeviceColor: 2
DeviceName: iPad
DieID: 5692515365953318
EthernetAddress: f0:76:6f:79:4f:fd
FirmwareVersion: iBoot-5540.120.17
HardwareModel: J71tAP
HardwarePlatform: s8003
HasSiDP: true
HostAttached: true
MLBSerialNumber: GCG74650QPQJ3QYAJ
ModelNumber: MP2G2
NonVolatileRAM: 
 auto-boot: dHJ1ZQ==
 backlight-level: MTU4NQ==
 boot-args: 
 com.apple.System.tz0-size: MHhFMDAwMDA=
PartitionType: GUID_partition_scheme
PasswordProtected: false
ProductName: iPhone OS
ProductType: iPad6,11
ProductVersion: 13.5.1
ProductionSOC: true
ProtocolVersion: 2
RegionInfo: FD/A
SBLockdownEverRegisteredKey: false
SerialNumber: GCGVPNVSHLFC
SoftwareBehavior: AQAAAAAAAAAAAAAAAAAAAA==
SoftwareBundleVersion: 
SupportedDeviceFamilies[2]: 
 0: 1
 1: 2
TelephonyCapability: false
TimeIntervalSince1970: 1595924985.687105
TimeZone: Europe/Warsaw
TimeZoneOffsetFromUTC: 7200.000000
TrustedHostAttached: true
UniqueChipID: 5692515365953318
UniqueDeviceID: 925253b6b76922294a4235ba2dafcd9e495ea3a7
UseRaptorCerts: true
Uses24HourClock: false
WiFiAddress: f0:76:6f:79:4f:fc
WirelessBoardSerialNumber: F20928D46C```

mexmer, 2020-07-28:
i would say apple broke it. tried with apple diagnostic (in osX) and show active no too

mexmer, 2020-07-30:
just to add to this tested on iOS 9, iOS12, iOS 13.5 an 13.6.
9 and 12 show wifi active yes
13 show wifi active no
all devices connected to same wifi.

nikias, 2020-08-04:
Apple definitely broke this. From syslog:
```
Aug  4 02:51:42 mobile_diagnostics_relay[5663] <Notice>: gather_nand_data: Could not get matching service for AppleNANDFTL
Aug  4 02:51:42 mobile_diagnostics_relay[5663] <Notice>: gather_nand_info: Could not gather the NAND information
Aug  4 02:51:42 mobile_diagnostics_relay[5663] <Notice>: gather_wifi_info: Could not find the IO80211Interface service
Aug  4 02:51:42 mobile_diagnostics_relay[5663] <Notice>: gather_hdmi_info: Could not find the AppleANX9836 service
```
Not much we can do here...