libimobiledevice/libimobiledevice #996: Issue with turning encryption on|off

tokenwizard, 2020-08-04:
So, I have a freshly reset device that I have never connected to iTunes and encryption was still OFF.

I'm just testing here, so I ran:
`idevicebackup2 encryption on --password 123456`

I then successfully took a full encrypted backup with:
`idevicebackup2 backup --full --password 123456 ~/iphone_backups/`

Lastly, I wanted to turn this temporary testing encryption back off, so I ran:
`idevicebackup2 encryption off --password 123456` 

But I get the following result/error:
_Started "com.apple.mobilebackup2" service on port 49645.
Negotiated Protocol Version 2.1
ErrorCode 207: Invalid password (MBErrorDomain/207)
Could not disable backup encryption._

I know this is the correct password, because I was able to successfully take the full backup and extract it using a third-party tool, using this password.

tokenwizard, 2020-08-04:
UPDATE:
If I run interactively and enter the password, it does work fine:
`idevicebackup2 -i encryption off
Started "com.apple.mobilebackup2" service on port 49676.
Negotiated Protocol Version 2.1
Enter current backup password: ******
Backup encryption has been disabled successfully.`

nikias, 2020-08-05:
@tokenwizard the `--password` option is not for enabling or disabling the password but to supply it when restoring an encrypted backup. To enable or disable the password by supplying it on the command line in non-interactive mode you add it as a parameter. See `idevicebackup2 --help`:
```
  encryption on|off [PWD]	enable or disable backup encryption
    NOTE: password will be requested in interactive mode if omitted
```

sfsantyf, 2022-09-20:
When I runned `$ idevicebackup2 --interactive --debug encryption off`  I got a prompt on my iDevice asking for authentication with the same error of this thread though **the passcode was valid**:
```
Negotiated Protocol Version 2.1
Please confirm disabling the backup encryption by entering the passcode on the device.
ErrorCode 207: Invalid password (MBErrorDomain/207)
Could not disable backup encryption.
```
For reasons that I don't know I solved it running the command as **superuser** rather than **_normal user_**:
```
Negotiated Protocol Version 2.1
Enter current backup password: ********
Please confirm disabling the backup encryption by entering the passcode on the device.
Backup encryption has been disabled successfully.
```
The last time that I did a backup was encrypted from iTunes, with idevicebackup2 seems like that didn't work with that last configuration and ended up in not asking the backup password but anyways it did it as superuser. 

I'm using `idevicebackup 1.3.0-165-gef7cf8e`, this may also be helpful for #1155.

nikias, 2022-09-20:
Oh hmm so it doesn't ask for the password unless you run it as root?!

sfsantyf, 2022-09-20:
@nikias I don't think so. I tried to enable and disable encryption for backups as normal user, this time my previous issue did not occur which means it's possible to enable or disable encryption for backups as normal user. The behavior that I had previously is probably not expected.

mexmer, 2022-09-21:
there should not be required elevation for password, since operation is done on device, not in your system, so as long, as you can communicate with device (which unprivileged user can), there should not be problem. from your snip it looks more like if something screwed command input, maybe some extra character after **off** from terminal or clipboard.