libimobiledevice/libimobiledevice #998: mobilebackup2 version exchange issue with iOS14beta4

zmdyy0318, 2020-08-06:
my device is iPhoneX iOS14beta4 and iPhone XR iOS14beta4.
when i use mobilebackup2_client_new(), it return MOBILEBACKUP2_E_BAD_VERSION.
then i find device_link_service_version_exchange(),iPhone return 
`["DLMessageVersionExchange", 400, 0]`
but 
`#define MBACKUP2_VERSION_INT1 300`
400 > 300,return with version judgment

zmdyy0318, 2020-08-06:
here is idevicebackup2 log:
```
02:35:05 property_list_service.c:253 internal_plist_receive_timeout(): printing 279 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
	<string>DLMessageVersionExchange</string>
	<integer>400</integer>
	<integer>0</integer>
</array>
</plist>
02:35:05 device_link_service.c:212 device_link_service_version_exchange(): Version mismatch: device=(400,0) > expected=(300,0)
02:35:05 mobilebackup2.c:89 mobilebackup2_client_new(): version exchange failed, error -6
```

nikias, 2020-08-06:
Fixed with commit b5d575c118ecfc2afcb12739433e916527182327.

closed 2020-08-06 by nikias

Dan-Maor, 2020-08-09:
This one is required for idevicescreenshot as well.

mexmer, 2020-08-10:
tbh. in our software, instead of sending fixed link version, we send back to device version it send to us.

during version exchange packet, first device sends you version (either it's expected version, or device link version), so we send same version back.

zmdyy0318, 2020-08-17:
This one is required for mobilesync as well. @nikias Thanks