libimobiledevice/libimobiledevice #1003: iOS 14 Support ?

darkzealot77, 2020-08-12:
Hi, I have installed the ios14 beta on a iphone 7 and try to do a backup on it :

_Started "com.apple.mobilebackup2" service on port 49253.
Could not perform backup protocol version exchange, error code -1_

It seems that lib can't exchange the version protocol, so it means that libeimobiledevice can't support ios14 at this time ?
Should I do something or do we have to wait a support of ios14 ?

mexmer, 2020-08-12:
yes, see #1001

darkzealot77, 2020-08-12:
Oh really ? Thank you for your response !

closed 2020-08-12 by darkzealot77

mastertinner, 2020-11-19:
@mexmer, will this be released soon? iOS 14 has been out for quite a while and it would be great if it was supported by an official release of libimobiledevice.

mexmer, 2020-11-19:
if by release you mean, when it will be available on different linux distributions, then it's up to linux distribution maintainers, tags for version with ios 14 support have been already made on repository.

mastertinner, 2020-11-19:
So should it be included in the `1.3.0` release? I'm using `1.3.0` on Arch Linux and am still getting the above error with an iOS 14 device.

nikias, 2020-11-19:
1.3.0 tag was pushed in June, this issue was fixed on August 7. So no, it's not tagged in a release yet. We will push a new tag soon.

mastertinner, 2020-11-19:
That's what I thought. Thanks, @nikias!

mexmer, 2020-11-19:
hmm, ok my bad than, i though support for 14 is already in 1.3, i know there were some fixes added later but for basic functionality ... guess not enough then.

maybe-sybr, 2021-01-13:
Hi @mexmer - just wanted to follow up on this to see if you might be able to tag a 1.3.1 which includes the iOS 14 fixes some time soon.

mastertinner, 2021-01-13:
> Hi @mexmer - just wanted to follow up on this to see if you might be able to tag a 1.3.1 (or 1.4.0) which includes the iOS 14 fixes some time soon.

Also tagging @nikias for that question :)

DrMWNH, 2022-01-02:
I'm getting the same error on OpenSuse Leap 15.2, with an iPhone SE running iOS 15.2.  Do the new iOS versions always deliberately break libimobiledevice?  If so, how long has it usually taken for the lib team to catch up?  thanks for any info.

mexmer, 2022-01-03:
@DrMWNH yes and and no. it's not that apple breaks libimobile library, they just change communication protocol parts, but sometimes those details are not backwards compatibile. in your case tho' problem would be libimobile binary shipped with your distribution. 
Stable distributions usually don't include binaries from experimental software, and last stable version of libimobile is https://github.com/libimobiledevice/libimobiledevice/releases/tag/1.3.0
which does not include, support for ios 14 and later.
you need to compile latest from sourcecode on github, if you want support for latest ios versions.

DrMWNH, 2022-01-04:
Josef, I'm extremely grateful for your personal reply and information.
I'll have another look at the documentation for the latest version to check
what is supported.
Many thanks for all your hard work creating these tools.
regards,
Miles

nikias, 2022-01-05:
@DrMWNH the only two things that broke back then was backup and screenshot capabilities. Everything else is still working with the latest stable release. However there is constant development happening so that it is always recommended to just compile the latest code from git to have the latest and greatest.