libimobiledevice/libimobiledevice #1004: idevicedebug could not start to com.apple.debugserver on iOS 14.0 beta4 devices under Xcode 12.0 beta4

albertdai, 2020-08-13:
How to reproduce it
1. Use `xcode-select -s` to switch to Xcode 12.0 beta4.
2. Run `idevicediagnostics restart` to restart the device.
3. Run `idevicedebug run` to launch the application.
4. Failed with `Could not start com.apple.debugserver!`.
5. Switching to Xcode 12.0 beta3 and repeating step 2, 3 could succeed.

Btw, I could still use Xcode IDE to launch application and use `xcodebuild` to launch test on the same device under Xcode 12.0 beta4.

MLKrisJohnson, 2020-08-14:
The problem is that the `com.apple.debugserver` service is no longer present as of Xcode 12 beta 4. The libimobiledevice code needs to use "com.apple.debugserver.DVTSecureSocketProxy" instead.

nikias, 2020-08-14:
For now, only 13.6 has the additional new service variant. The basic approach here is to try the new one and if it fails try the old one.

albertdai, 2020-08-14:
Thanks for finding that. Any ETA for fixing in the master code?

albertdai, 2020-08-20:
I have verified locally that replacing `com.apple.debugserver` with `com.apple.debugserver.DVTSecureSocketProxy` could not fix the issue.

And also no luck on Xcode 12.0 beta5.

albertdai, 2020-08-23:
Thanks Kris. I could confirm it could fix the issue.

albertdai, 2020-08-25:
The structure `idevice_t` has the member `version`. We could use this member to determine if using the new service name or not.

nikias, 2020-09-10:
I pushed commit 98056a89648f431759c5fa4ed87c6ea6ba0cdd3f which solves it for debugserver. For any other service it has to be handled separately like I did with debugserver.

closed 2020-09-10 by nikias