libimobiledevice/libimobiledevice #1005: Unbacking a backup seems to require the device to be connected

gabrc52, 2020-08-16:
It seems that using `idevicebackup2 unback .` requires the device to be connected in order to unback, even if you specify the UUID.

Is this restriction necessary? It seems everything needed to unback is in the subdirectories and Manifest.db ("actual" file locations per unique file ID).

nikias, 2020-08-16:
This is how the operation is implemented. Good luck unbacking an encrypted back on your own. However, sadly, Apple broke the unback command a few major iOS versions back.

pabs3, 2021-04-27:
For an unencrypted backup it should be possible though right?

PS: on the iPad I am looking at, the manifest file is Manifest.mbdb.

pabs3, 2021-04-27:
I just tried the [ideviceunback](https://github.com/inflex/ideviceunback) tool mentioned in #528 and it is a good replacement for this missing feature.

@inflex I expect the ideviceunback code could just be added to `idevicebackup2 unback` behind a `--offline` option, have you considered submitting a pull request for this feature?