libimobiledevice/libimobiledevice #1007: idevicebackup2 Restore - What command/flag/specific code makes the device to trigger the "press home to upgrade" when rebooting

bricedenice59, 2020-08-17:
Hi guys, I have a custom implementation of the restore part in C#, it works well, can restore whatever backup, but I have not the same behavior at the very end of the process. Once it is finished, it reboots but my version never triggers the white screen "press home to upgrade". I played with all the possible settings to try replicate like your binary. I just can't.
With my version, after rebooting the sms app shows no message, it is just empty, if I use same backup and your binary to restore again, sms are back on device!
So I really suspect important things are processed after "press home to upgrade" as I can see it takes time for the progress bar to go to 100%.

They are the flags I pass to the SendRequest
```
            opts["RestoreSystemFiles"] = true;
            opts["RestoreShouldReboot"] = true;
            opts["RestoreDontCopyBackup"] = true;
            opts["RestorePreserveSettings"] = true;
            opts["RemoveItemsNotRestored"] = false;
```
If you have any idea that may help me :) 
Thanks a lot in advance.

nikias, 2020-08-17:
Afaik, the "Press home to upgrade" screen is displayed whenever you restore a backup with a lower version than the one installed on the device. There is definitely no option that would trigger it.

bricedenice59, 2020-08-18:
Finally I found that passing flags = true to RemoveItemsNotRestored triggers the "Press home to upgrade" .
It is fine now 👍

closed 2020-08-18 by bricedenice59

nikias, 2020-08-18:
Oh, interesting find, thanks for the feedback.