libimobiledevice/libimobiledevice #1013: iPhone iOS 14 Backup failed.

hiephitek, 2020-08-24:
Hello Dev team,
I am trying to backup my iPhone 6S - iOS 14 then i get the output:
```
$ idevicebackup2 backup --full ./
Backup directory is "./"
Started "com.apple.mobilebackup2" service on port 52638.
Could not perform backup protocol version exchange, error code -1
```
A backup works with iTunes, So how can it work with idevicebackup2?
Thanks for your great work.

mexmer, 2020-08-24:
make sure you are using latest version from repository. ios 14 backup has been already fixed, but if you get version exchange error, it suggest you using old version of libimobile.

hiephitek, 2020-08-24:
> make sure you are using latest version from repository. ios 14 backup has been already fixed, but if you get version exchange error, it suggest you using old version of libimobile.

Sure, I am using the latest version from the repository.

mexmer, 2020-08-25:
recompile libimobile with --enable-debug, then you will be able to see communication with mobilebackup service, and post here part, where it try to exchange version.

nikias, 2020-08-25:
@hiephitek with that error message you can be sure that you're not using the latest code. Make sure your idevicebackup2 isn't linked against some older libimobiledevice.

hiephitek, 2020-08-25:
@nikias , @mexmer 
It was my mistake. Now i can backup my iPhone.
Thanks for your support.

closed 2020-08-25 by hiephitek

brtvrdrl, 2020-09-18:
i have same issue, is it possible to give me compiled new version of idevicebackup2.exe work with iOS 14 please. Im not good at programming but i need it. Please give me link that i can download directly. Thanks

brtvrdrl, 2020-09-18:
> @hiephitek with that error message you can be sure that you're not using the latest code. Make sure your idevicebackup2 isn't linked against some older libimobiledevice.

i have same issue, is it possible to give me compiled new version of idevicebackup2.exe work with iOS 14 please. Im not good at programming but i need it. Please give me link that i can download directly. Thanks

LukeLR, 2020-12-19:
I have the same issue on Arch Linux. `idevicebackup2 -v` says `idevicebackup2 1.3.0`, which appears to be the latest version. However, when trying to start a backup of my iOS 14 iPad, I still get the error stated above. When building `libimobiledevice` from git, `idevicebackup2 -v` says `idevicebackup2 1.3.1` and now it seems to work. Any ideas?

mexmer, 2020-12-20:
either manually build own version, or wait until arch linux updates packages

web-refinery, 2021-05-16:
Same issue with arch here. Solution: install the [AUR: libimobiledevice-git](https://aur.archlinux.org/packages/libimobiledevice-git/) version which features version 1.3.1.

cedricloneux, 2021-08-31:
why not release version 1.3.1 ? Last release is 1.3.0, same problem with homebrew