libimobiledevice/libimobiledevice #1015: idevicescreenshot fails on iOS 14 beta 5: Could not connect to screenshotr!

jmagman, 2020-08-26:
```
idevicescreenshot test.png --udid 00008027-001925360106802E
Could not connect to screenshotr!
```

Works on iOS 13.5.1.

jmagman, 2020-08-26:
Did the service name change?

nikias, 2020-08-26:
The protocol did change, make sure to update libimobiledevice to latest git code

jmagman, 2020-08-26:
Should have mentioned I was on idevicescreenshot 1.3.0.  Thanks, will upgrade and try again!

closed 2020-08-26 by jmagman

yatouxingren, 2021-04-23:
How to solve this problem？my idevicescreenshot version is 1.3.0，but the problem remains

ilyalychkou, 2021-08-07:
> How to solve this problem？my idevicescreenshot version is 1.3.0，but the problem remains
the same issue.

nikias, 2021-08-07:
1.3.0 is not latest. You need to rebuild from source to get the latest code.

Ramia111, 2022-07-31:
> 1.3.0 is not latest. You need to rebuild from source to get the latest code.

hello, according to https://github.com/libimobiledevice/libimobiledevice/releases, 1.3.0 is the latest. I am confused. Could you please provide the webpage of the latest code?

mexmer, 2022-08-01:
@Ramia111 releases are release packages, while latest code is in repository, clone repo and built from code, don't use prepackaged version, which from release date, you can see, is outdated

bensh, 2023-03-13:
Found this on another bug feed. Someone made a build script.
Kudos to them, worked well for me

mkdir -p limd-build
cd limd-build
curl -o ./limd-build-macos.sh -Ls https://git.io/JEhrV 
bash ./limd-build-macos.sh

mexmer, 2023-03-13:
@bensh that someone is nikias

nikias, 2023-03-13:
@bensh @mexmer 🤣