libimobiledevice/libimobiledevice #1018: Exit recovery?

MrBenFTW, 2020-08-27:
Hi, I noticed you can make the device enter recovery mode with `ideviceenterrecovery` however, I cannot figure out how to make it exit recovery through command line. 

Do you know how I would go about doing that?

Also, is it possible to send the device to DFU mode rather than recovery and also exit?

nikias, 2020-08-27:
Use the `irecovery` tool from `libirecovery` (https://github.com/libimobiledevice/libirecovery) and then run
```
irecovery -n
```

MrBenFTW, 2020-08-27:
Thank you, I'm on macOS though, are there any compiled releases? I've tried installing dependencies myself but I'm never successful

MrBenFTW, 2020-08-27:
Never mind, figured out what I was doing wrong :)

MrBenFTW, 2020-08-27:
Is entering DFU mode possible?

nikias, 2020-08-27:
Not programmatically.

MrBenFTW, 2020-08-27:
Okay thanks, I am trying to make a simple app which has a "enter recovery" and "exit recovery" button but I can't figure out how to embed irecovery inside the project and then run it

nikias, 2020-08-27:
Not sure why you need an app for that but you should link against libimobiledevice and libirecovery and have your code call the appropriate API functions. But this is out of scope of this ticket. Check the code in ideviceenterrecovery and irecovery how it works.

indis-anarion, 2025-06-21:
> Use the `irecovery` tool from `libirecovery` (https://github.com/libimobiledevice/libirecovery) and then run
> 
> ```
> irecovery -n
> ```

it's working. Thanks you

MisutaaAsriel, 2026-01-30:
> Never mind, figured out what I was doing wrong :)

~~Uhm can you share with the rest of the class?~~

For those who got here on a web search and are running macOS, if you have homebrew (`brew`) installed, you can run:
```
brew install libirecovery
```

Followed by:
```
irecovery -n
```

(With the device connected of course.)