libimobiledevice/libimobiledevice #1019: Instructions to install on Ubuntu 20.04 Focal Fossa

victoriadrake, 2020-08-29:
I believe the instructions at https://libimobiledevice.org/#get-started for Ubuntu are outdated. To successfully install, I ran:

```bash
sudo apt install libimobiledevice6 usbmuxd libimobiledevice-utils
```

Only afterwards was I able to successfully see device info with `ideviceinfo`.

andrei91ro, 2020-10-29:
I can confirm that these are the correct commands to install on Ubuntu 20.04.
These should appear on the front page, before any compilation instructions

JunCEEE, 2021-05-23:
Yes, I think victoriadrake is correct. On this [page](https://libimobiledevice.org/#get-started) for ubuntu, the `libimobiledevice` cannot be found. I guess it should be replaced by `libimobiledevice6`. Maybe someone can get the page updated? @nikias @FunkyM Many thanks.
[image]

nikias, 2021-05-23:
Updated the page to say `libimobiledevice6`.

closed 2021-05-23 by nikias