libimobiledevice/libimobiledevice #1020: Does the C library provide more data than the provided command line scripts?

ScottPierce, 2020-08-30:
The data is extremely accessible with the provided command line scripts, but there is certain data that I'm looking for from the iOS device, but not seeing. I'm also not a C engineer, so using C and relying on it are foreign concepts for me. Is there more data provided via the C library, or is all of the data provided by the command line scripts?

XanaduAPI, 2020-09-12:
@ScottPierce 
Using C / C + + to call this module can get more data.

XanaduAPI, 2020-09-12:
@ScottPierce 
auto	vClientDiagnostics = static_cast<diagnostics_relay_client_t>(nullptr);
auto	vError = diagnostics_relay_client_start_service(vClientDevice, &vClientDiagnostics, "diagnostics_relay_client");
if(DIAGNOSTICS_RELAY_E_SUCCESS == vError && vClientDiagnostics)
{
	//read battery info
	auto	vBatteryList = static_cast<plist_t>(nullptr);
	diagnostics_relay_query_ioregistry_entry(vClientDiagnostics, nullptr, "IOPMPowerSource", &vBatteryList);
	if(vBatteryList)
	{
		// printf plist
	}

	diagnostics_relay_client_free(vClientDiagnostics);
	vClientDiagnostics = nullptr;
}