libimobiledevice/libimobiledevice #1021: idevicedebug did not kill the running process

albertdai, 2020-08-31:
In the code of `idevicedebug`, it is expected that the tool could kill the process: https://github.com/libimobiledevice/libimobiledevice/blob/master/tools/idevicedebug.c#L489

Actually, it could not work for iOS 12 and 13.

But it is interesting that after patching some changes for iOS 14 (https://github.com/libimobiledevice/libimobiledevice/issues/1004), this ability comes back.

It looks the difference are the tool did not run `service_disable_bypass_ssl` when the tool starts the debugserver: https://github.com/libimobiledevice/libimobiledevice/blob/master/src/debugserver.c#L83 and the new service `com.apple.debugserver.DVTSecureSocketProxy` enables the SSL.

jpstotz, 2020-09-22:
I also noticed this change and it gives me quite a headache, because I want to exit idevicedebug but **not kill the app**. 

I already tried to disconnect the debugger by sending the `D` command and directly afterwards exit idevicedebug but the app still terminates.

star6r6, 2020-12-04:
Same here.
I want to exit idevicedebug but not kill the app.
Can idevicedebug give an option to disable this feature?

jpstotz, 2020-12-04:
@star6r6 I already prepared a pull request that introduces such an option: https://github.com/libimobiledevice/libimobiledevice/pull/1050

star6r6, 2020-12-07:
> @star6r6 I already prepared a pull request that introduces such an option: #1050
> 
> If I use the version from this PR I am able to start an app and disconnect on my iOS 14.2 device.

Thanks. I will try it later.

jpstotz, 2020-12-10:
@star6r6  My pull request has been accepted and merged into official master branch of libimobiledevice project. 
To use it just clone and compile it and you can use the new `--detach` option.

star6r6, 2020-12-15:
@jpstotz Already in use.Nice feature!

warpfactorfive, 2021-01-10:
I would like to use the new detach flag but I installed the library through debian apt-get.  I just checked, and it doesn't look like the apt central repository has been updated to include this change.  Any idea how often that happens, or how to request it?

nikias, 2021-01-11:
They will update once a new official version is released, which will happen in the near future. In the meantime, you have to compile it yourself.

jpstotz, 2021-01-11:
@warpfactorfive If you are on a Debian system then you should better build libimobiledevice (and it's dependent projects) yourself. If I get it correctly Debian repo is still at  version 1.2.1~git20181030 so they ignored v1.3 released July last year. Therefore the chance that the new version will be included soon is very low. 
Other Debian based Linux distributions like Ubuntu also don't update libimobiledevice for existing Ubuntu versions. On Ubuntu libimobiledevice is only updated for new releases of Ubuntu (but usually the used version is not the recent one, instead it is 4-12 months old).  

If you don't have software installed via apt/dpkg that has libimobiledevice, (lib)usbmuxd or libplist as dependency I would recommend you to to uninstall those packages and directly install the latest version from git. Unfortunately this means that you have to clone, build and install them all manually in the correct order (as far as I remember libplist, usbmuxd, libusbmuxd, libimobiledevice, ..).

warpfactorfive, 2021-01-11:
@jpstotz  thanks for the suggestion.  Will do as you suggest