libimobiledevice/libimobiledevice #1022: idevice_id -l command is not listing simulator device

yaseenmd, 2020-09-02:
I installed libimobiledevice through  brew install libimobiledevice. 

And i have launched iphone simulator on mac.  

idevice_id -l is not returning udid of simulator.

mexmer, 2020-09-03:
simulator is not communicating trough usbmux, therefore libimobiledevice (which asks usbmuxd about device), can't see it.

xcode sees emulator, because it launched it, and knows on which port it needs to communicate with it.

yaseenmd, 2020-09-03:
Is there any library available (like usbmuxd) for simulator device communication?

nikias, 2020-09-09:
No. The simulators work totally different. They don't have services, and the "apps" are not even iOS binaries but x86_64 and run directly in the operating system userspace.

chmiiller, 2022-08-12:
Actually there's a way to do it with `com.apple.iphonesimulator.simulateLocation` and there's also a Node module (with a CLI) that allows you to do so: https://github.com/steffenmllr/ios-simulator-set-location

mexmer, 2022-08-15:
if you check the code, you can see, it call simulator service ... which is (like @nikias said), something totally different, than communicating with device trough usbmuxd.

that code is sending notification to system service that manages running simulator devices