libimobiledevice/libimobiledevice #1026: SSL_read Crash

MikuApi, 2020-09-10:
In MacOS, use libimobile device-1.2.1. Crash will occur occasionally, and it is the same in windows, but it can be used__ try/__ except(EXCEPTION_ EXECUTE_ Handler) capture, but I didn't find it in MAC.

Here is the crash code

Thread 2 Crashed:
0   ???                           	000000000000000000 0 + 0
1   libcrypto.1.1.dylib           	0x00000001134dbca2 CRYPTO_gcm128_decrypt_ctr32 + 610
2   libcrypto.1.1.dylib           	0x00000001134b3768 aes_gcm_cipher + 824
3   libssl.1.1.dylib              	0x0000000113350bbc tls1_enc + 2060
4   libssl.1.1.dylib              	0x000000011334f2c8 ssl3_get_record + 1592
5   libssl.1.1.dylib              	0x000000011334dfc0 ssl3_read_bytes + 1696
6   libssl.1.1.dylib              	0x0000000113356d0c ssl3_read_internal + 188
7   libssl.1.1.dylib              	0x0000000113363878 SSL_read + 24
8   libimobiledevice.6.dylib      	0x0000000113c6f9b2 idevice_connection_receive_timeout + 162 (idevice.c:552)
9   libimobiledevice.6.dylib      	0x0000000113c70064 service_receive_with_timeout + 52 (service.c:189)
10  libimobiledevice.6.dylib      	0x0000000113c703ba internal_plist_receive_timeout + 106 (property_list_service.c:209)
11  libimobiledevice.6.dylib      	0x0000000113c7140e lockdownd_get_value + 286 (lockdown.c:230)
12  libAppleDevicePlatform.1.dylib	0x000000010906ee24 lockdownd_error_t FunctionCallLockdownd<lockdownd_error_t (*)(void*, char const*, char const*, void**), void*, char const*, char const*, void**>(lockdownd_error_t (*)(void*, char const*, char const*, void**), void*, char const*, char const*, void**) + 68 (AppleDeviceDriveTemplate.h:279)
13  libAppleDevicePlatform.1.dylib	0x000000010906edcb lockdownd_get_value(void*, char const*, char const*, void**) + 75 (AppleDriveLockdownd.cpp:113)
14  libAppleDevice.1.dylib        	0x0000000109191002 CDevice::CallBack_Heartbeat() + 98
15  libAppleDevice.1.dylib        	0x000000010919111c non-virtual thunk to CDevice::CallBack_Heartbeat() + 28
16  libAppleDevice.1.dylib        	0x0000000109259152 CDeviceManager::ThreadCallBackHeartbeat(void*) + 450 (CDeviceManager.cpp:761)
17  libAppleDevicePlatform.1.dylib	0x0000000109036915 Thread_Function_Windows(void*) + 53 (ADPlatformProcess.cpp:600)
18  libAppleDevicePlatform.1.dylib	0x0000000109036965 Thread_Function_Linux(void*) + 21 (ADPlatformProcess.cpp:607)
19  libAppleDevicePlatform.1.dylib	0x0000000109036985 Thread_Function_Mac(void*) + 21 (ADPlatformProcess.cpp:613)
20  libsystem_pthread.dylib       	0x00007fff9cdaf93b _pthread_body + 180
21  libsystem_pthread.dylib       	0x00007fff9cdaf887 _pthread_start + 286
22  libsystem_pthread.dylib       	0x00007fff9cdaf08d thread_start + 13

nikias, 2020-09-10:
This is not coming from the code in libimobiledevice itself, there's nothing we can fix here...

XanaduAPI, 2020-09-11:
Hello, lockdownd_ get_value() support concurrent calls?

I'm sorry, I'm using the translation tool.

https://github.com/MobiVM/robovm/blob/c8d71aa63481c1b03edfb99ec1130bd345d6764d/compiler/libimobiledevice/patches/04-fix-ssl_read_with_timeout.patch.

But with this patch, it seems to be getting better

closed 2020-09-12 by XanaduAPI