libimobiledevice/libimobiledevice #1028: idevicebackup2.exe gives error code -1 on iOS14

brtvrdrl, 2020-09-19:
here is the error message

Backup directory is "C:\VardarMobile\VardarMobile\backup"
Started "com.apple.mobilebackup2" service on port 49610.
Could not perform backup protocol version exchange, error code -1

is it possible to provide or compile updated idevicebackup2.exe for windows10-64bit pc for me.  Thanks

nikias, 2020-09-19:
You need to update libimobiledevice to latest from git to get support for iOS 14.

brtvrdrl, 2020-09-19:
my firend and me have tried to update it but we couldnt achieve it to compile unfortunately. But its needed urgent. Can you do it for me and give me download link please.  Thanks

brtvrdrl, 2020-09-20:
I have tried to compile it again but it gives some errors and it doesnt compile in the end. Please compile it for windows and share with me. it will be big help for everyone who got same issue as me. Thanks

YSL21, 2020-11-11:
Compile seemed to work but still got the same issue as of yesterday... any updates on a fix?

bobobo1618, 2021-01-19:
I'm able to successfully compile and pull a backup on Linux using the latest Git version.

@brtvrdrl @YSL21 I don't think many of the people around here run Windows but it looks like [this project](https://github.com/libimobiledevice-win32/imobiledevice-net) provides development builds [here](https://dev.azure.com/libimobiledevice-win32/imobiledevice-net/_build/latest?definitionId=1&branchName=master).

mastertinner, 2021-08-06:
@nikias, would it make sense to release a new version of `libimobiledevice` that supports iOS 14? It has been almost a year now since iOS 14 was released, so it would be great to get official released support from this library for it.

nikias, 2021-08-06:
@mastertinner I am currently restructuring some things (e.g. moving common code to a new library [libimobiledevice-glue](https://github.com/libimobiledevice/libimobiledevice-glue)) and some other remaining things, and then we will have a new release.