libimobiledevice/libimobiledevice #1033: Skip Media while taking backup

sKerry-not-scarry, 2020-09-21:
Is there any way in which we can skip media while taking backup?
I tried a way in which I was able to skip the file to be written on the system but it still takes up time to read (I guess?) from the iPhone. I am looking to reduce that time by skipping reading as well (if it is reading, otherwise please help.).

I was able to reduce the file size of backup by a large margin. (If anyone wants the code, feel free to ask, I am new to this but you can try the code if it works for you)

My main aim is to take backup of a particular file only and then restore. If I do the complete backup it takes time and I want to reduce the time.

mexmer, 2020-09-22:
no, backup is completely under control of phone. also if you delete files from backup, you will not be able to restore that backup.

sKerry-not-scarry, 2020-09-22:
Thanks for commenting @mexmer .
I actually want to alter a value in a plist file. To be more specific I want to put the iPhone under supervision. Is there any other way to do it.

mexmer, 2020-09-23:
i'm not exactly sure what you mean by supervision, there is device management and provisioning profiles, but for that you actually need mac, because configuration profiles can be created only by apple configurator, which is available only for OS X

sKerry-not-scarry, 2020-09-23:
what you said is absolutely correct, what I am looking for is to put the iPhone under supervision. 
There a plist which contains an `IsSupervised` flag which is responsible for the same. (Screenshot attached)
Right now I back up the whole device, change the flag and then restore and voilà, the device goes in and out of supervision. 
Now this process is very lengthy as it requires the whole iPhone to be backed up and then restored. 
I just need this process to shorten.

closed 2020-10-08 by sKerry-not-scarry