libimobiledevice/libimobiledevice #1035: ideviceinfo 1.3.0 and other utilities cannot find device

launchdaemon, 2020-09-22:
ideviceinfo, idevice_id and other tools from the library all return "ERROR: Unable to retrieve device list!".

I am running Ubuntu 18.04.5 LTS and have the latest version of libimobiledevice, 1.3.0, which was installed by apt when updating checkra1n.

I have tried with an iPhone 5S on 12.4.3, another on 12.4.8 and an iPhone 5 on 10.3.3.  I connect the device, ensure it is unlocked and the host trusted, but for all three devices all of the utils respond with variations of the above message.

for ideviceinfo, -d don't provide and different output and -s doesn't behave any differently.  Likewise using -l or -d for idevice_id.

Using sudo does not make a difference.

The device is seen by lsusb and is mounted by the file explorer to access pictures etc.

nikias, 2020-09-22:
You need to make sure you have usbmuxd installed and set up.

launchdaemon, 2020-09-22:
Hi @nikias 

Firstly, thank you for the quick reply and for all your hard work on these incredibly useful tools.

Sorry, I should have checked ubsmuxd before logging the issue. Until updating checkra1n (and along with it libimobiledevice) I had been able to to use tools such as ideviceinfo to query a device or iproxy to access it.

It appears to be a problem with usbmuxd crashing when I connect a device or if I restart usbmuxd with a device already connected.

Without a device connected it appears to start fine:

```
$ sudo /usr/sbin/usbmuxd -vv --user usbmux --systemd
[16:20:31.579][3] usbmuxd v1.1.0 starting up
[16:20:31.579][4] Creating socket
[16:20:31.580][5] initialized config_dir to /var/lib/lockdown
[16:20:31.582][3] Successfully dropped privileges to 'usbmux'
[16:20:31.582][5] client_init
[16:20:31.582][5] device_init
[16:20:31.582][4] Initializing USB
[16:20:31.582][5] usb_init for linux / libusb 1.0
[16:20:31.593][4] 0 devices detected
[16:20:31.593][3] Initialization complete
[16:20:31.593][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.
[16:20:31.593][4] New client on fd 11
[16:20:31.593][5] Client command in fd 11 len 435 ver 1 msg 8 tag 8
[16:20:31.593][5] send_pkt fd 11 tag 8 msg 8 payload_length 278
[16:20:31.593][5] Client 11 now LISTENING
```
but on attaching a device get the following:
```
[16:21:15.656][4] New client on fd 12
[16:21:15.656][5] Client command in fd 12 len 432 ver 1 msg 8 tag 1
[16:21:15.656][5] send_pkt fd 12 tag 1 msg 8 payload_length 221
[16:21:15.656][4] Client 12 connection closed
[16:21:15.656][4] Disconnecting client fd 12
[16:21:16.644][4] Found new device with v/p 05ac:12a8 at 1-9
[16:21:16.644][3] Could not get old configuration descriptor for device 1-9: -5
[16:21:16.644][4] Setting configuration for device 1-9, from 0 to 4
[16:21:16.663][4] Found interface 1 with endpoints 04/85 for device 1-9
[16:21:16.664][4] Using wMaxPacketSize=512 for device 1-9
[16:21:16.664][3] Connecting to new device on location 0x10009 as ID 1
[16:21:16.664][5] All 3 RX loops started successfully
[16:21:16.665][3] Connected to v2.0 device 1 on location 0x10009 with serial number XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
[16:21:16.665][4] preflight_worker_handle_device_add: Starting preflight on device XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX...
[16:21:16.665][1] preflight_worker_handle_device_add: ERROR: Could not connect to lockdownd on device XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX, lockdown error -8
```
I realise it is not the latest 1.1.1 version.  That was not available via apt but can try and build it if you think that will fix the issue.

Many thanks

launchdaemon, 2020-09-23:
Updating usbmuxd to the version just released via apt fixed the issue.

Thanks again.

closed 2020-09-23 by launchdaemon