libimobiledevice/libimobiledevice #1037: iOS 13.6/13.7/14.0 app installation + device reboot often end up in "Could not connect to lockdownd, error code -8"

jpstotz, 2020-09-23:
Starting with iOS 13.6 there seems to be a problem that causes a complete failure of the communication to an iOS device if the device reboots:

If I execute the following commands in a script this ends up with a chance in a total loss of communication between PC and iPhone:
 
```
#!/bin/bash
for i in {1..10}
do
   ideviceinstaller -i some-ios-app.ipa
   idevicediagnostics restart
   echo "Loop $i"
   sleep 40s
done
```
Usually after running the loop about 3-6 times I end up with the error:
```
Could not connect to lockdownd. Exiting.
ERROR: Could not connect to lockdownd, error code -8
``` 
The only way to end up this error situation is to disconnect and reconnect the USB connection between PC and iPhone.
 
Tested using an iPhone X on iOS14 + Ubuntu 20.04 using HEAD versions of libimobiledevice + it's libraries.

Note the same scripts works (based on my observation) flawless for iOS devices running e.g. iOS 13.5.1 or 12.4.

jpstotz, 2020-09-25:
I performed the test described in my first post on a second computer, this time an old mac-mini running Catalina 10.15.7 and again HEAD versions of libimobiledevice, ideviceinstaller, usbmuxd and libplist. The used iOS14 device stays the same.

In this test the problem does not occur, which means that there are two potential reasons that may cause this problem: 

1. The timing. A different OS with an older CPU may totally change the timing of some commands, especially if the are executed concurrently
2. Some part of the connection works different on iOS than on Ubuntu 20.04

jpstotz, 2020-10-06:
I made a third test on a Raspberry using latest RaspbianOS headless. Again the test worked without problems. 

Therefore it seems like this problem is really on iOS side and not on libimobiledevice side. I assume that it is some sort of race condition in the lockdownd on iOS caused by Ubuntu or Gnome or any other software that automatically connects to detected iOS devices via usbmuxd. 

Does anybody know if there is an easy way to trace all connections and data going through usbmuxd? I have already tested the verbose mode but that is not enough to understand what is going wrong.

jpstotz, 2020-10-21:
I made some more tests and found out that the reason why the test-script works on some devices and fails on others is the Gnome Virtual File System (gvfs). 

If gvfs is installed and running on a system the the error occurs with a chance of may be 30-50% per device reboot. 

If gvfs is not installed or all gvs processes has been stopped the test script works again without any problems.

qmfrederik, 2020-12-04:
Yes, I can confirm this. I ended up uninstalling `gvfs-backends` alltogether.